aseba-community/aseba #709: https://www.thymio.org/en:linuxinstall Others - Page not found

paolorossetti, 2017-10-22:
I am trying to install Aseba on So.Di.Linux the official Linux Distribution for Italian Schools by CNR.

Since the instructions for UBUNTU does not work (error reported is "NO JSON object found") I went down the  https://www.thymio.org/en:linuxinstall page till "Others" and in the link
"compile Aseba from source." I get a null page. Either in English, French and Italian pages... 

https://www.thymio.org/en:sourceinstall
https://www.thymio.org/fr:sourceinstall
https://www.thymio.org/it:sourceinstall

report to a missing page. 

How to proceed? I would like to try to install Aseba on So.Di.Linux... 

Thanks for any advice.

mbonani, 2017-10-23:
Hi,

The links are corrected, the correct info could be reach from the
https://www.thymio.org/en:start pages on the "source code " icon.

Regards

Michael Bonani

Association Mobsya
Ch. du Closel 3
1020 Renens

Tél: +41 (0)32 511 64 93

closed 2017-10-23 by mbonani

stephanemagnenat, 2017-10-23:
Hi Paolo. From https://it.wikipedia.org/wiki/So.Di.Linux I see that So.Di.Linux is an Ubuntu derivative, the 2016 version should be compatible with Ubuntu 16.04. In that respect, the Ubuntu instructions should work. Could you please give more information on how/when you got the "NO JSON object found" error? Thank you.

paolorossetti, 2017-10-23:
Ciao Stéphane,

I tried to add the PPA with the instruction found here https://www.thymio.org/en:linuxinstall <https://www.thymio.org/en:linuxinstall>

sudo add-apt-repository ppa:stephane.magnenat/`lsb_release -c -s`
and I get this error on SoDiLinux@CTS 2015

Impossible to add PPA "NO JSON OBJECT Could be decoded".

Last Saturday I did a 4 hour training in a school to prepare for the next R2T2 to 15 kids. Out of 8 PC with Windows XP and Windows 10 only 3 were good enough to run VPL (I guess that they are too obsolete for 1.5.5) Last year they were working with the former Aseba but this year I had many problems. Therefore I planned to help the school to migrate to So.Di.Linux that in Italy is accepted since it contains the INVALSI test sw (that usually runs only on Windows) that is mandatory for the EU testing schema. And also several SW for kids with disabilities. 

I will ask Giovanni Caruso to let me know if he had the same problem as me.

Thanks for your support!
Paolo

> Hi Paolo. From https://it.wikipedia.org/wiki/So.Di.Linux <https://it.wikipedia.org/wiki/So.Di.Linux> I see that So.Di.Linux is an Ubuntu derivative, the 2016 version should be compatible with Ubuntu [...]

Ing. Paolo Rossetti

Committee Member
Association MOBSYA – Robots au service de l'éducation 
Chemin du Closel 5 – 1020 Renens – Lausanne – Suisse
paolo.rossetti@mobsya.org
+39 335 6968 980

stephanemagnenat, 2017-10-23:
I see, the problem is that the script assumes Ubuntu, while your distribution, although based on Ubuntu, changed its name.

Assuming you have the latest version of So.Di.Linux, you can try:
```
sudo add-apt-repository ppa:stephane.magnenat/xenial
sudo apt-get update
sudo apt-get install aseba
sudo adduser $USER dialout
```
Please tell me whether it works or not.

thank you :)

paolorossetti, 2017-10-23:
I will try in a minute and let you know!

THANKS!
> On 23 Oct 2017, at 11:32, Stéphane Magnenat <notifications@github.com> wrote:
> 
> sudo add-apt-repository ppa:stephane.magnenat/xenial
> sudo apt-get update
> sudo apt-get install aseba
> sudo adduser $USER dialout

Ing. Paolo Rossetti

Committee Member
Association MOBSYA – Robots au service de l'éducation 
Chemin du Closel 5 – 1020 Renens – Lausanne – Suisse
paolo.rossetti@mobsya.org
+39 335 6968 980