aseba-community/aseba #716: Error when building aseba.*.deb

ypiguet-epfl, 2017-11-09:
Following the [build instructions](https://github.com/aseba-community/aseba/blob/master/compile.Linux.md) on Ubuntu 16.04 yields the error below when running `debuild -i -us -uc -b` for aseba:
```
CMake Error at /usr/share/dashel/CMake/dashelConfig.cmake:48 (find_package_handle_standard_args):
  Unknown CMake command "find_package_handle_standard_args".
Call Stack (most recent call first):
  CMakeLists.txt:64 (find_package)
-- Configuring incomplete, errors occurred!
```

ypiguet-epfl, 2017-11-14:
Patch suggestion:
```
--- dashel/debian/build/dashelConfig.cmake-orig	2017-11-14 14:46:51.857803928 +0100
+++ dashel/debian/build/dashelConfig.cmake	2017-11-14 14:46:59.317789228 +0100
@@ -45,4 +45,5 @@
 find_path(dashel_INCLUDE_DIRS dashel/dashel.h /home/piguet/work/aseba-20171108/dashel CMAKE_FIND_ROOT_PATH_BOTH)
 find_library(dashel_LIBRARY dashel /home/piguet/work/aseba-20171108/dashel/debian/build CMAKE_FIND_ROOT_PATH_BOTH)
 set(dashel_LIBRARIES ${dashel_LIBRARY} ${EXTRA_LIBS})
+include(FindPackageHandleStandardArgs)
 find_package_handle_standard_args(dashel DEFAULT_MSG dashel_LIBRARIES dashel_INCLUDE_DIRS)
```

Then debuild complains about an existing directory:
```
mkdir debian/build
mkdir: cannot create directory ‘debian/build’: File exists
debian/rules:7: recipe for target 'build' failed
```
Patch suggestion:
```
--- aseba/debian/rules-orig	2017-11-14 14:39:07.922535469 +0100
+++ aseba/debian/rules	2017-11-14 14:39:12.330530807 +0100
@@ -4,7 +4,7 @@
 VERSION := $(shell echo '$(DEBVERS)' | sed -e 's/^[[:digit:]]*://' -e 's/[~-].*//')
 
 build:
-	mkdir debian/build
+	mkdir -p debian/build
 	cd debian/build && cmake -D USER_BUILD_VERSION=${DEBVERS} -D CMAKE_BUILD_TYPE=Release -D CMAKE_INSTALL_PREFIX=../tmp/usr ../..
 	cd debian/build && $(MAKE)
```

stephanemagnenat, 2017-11-14:
Patches applied in aseba-community/dashel@bbfe0e9256ac05bb9a001e7af1b76ad8f4152120 and in 517cb2709e14bb8c761795eaceada1007aaee0cc. Please test and close if it works for you.

Thank you for reporting and suggesting patches!

ypiguet-epfl, 2017-11-21:
Tested in Ubuntu 16.04 and docker, works fine. Thanks!

closed 2017-11-21 by stephanemagnenat