aseba-community/aseba #719: unifr.playground asebaswitch zeroconf

mbonani, 2017-11-17:
The unifr as in addition a asebaswitch on port 33333. 
This is for make possible to connect all 4 epucks, but it make strange behavior, like on computer can connect to several robot or not. 
I suggest to remove this switch on the example or update actual asebaswitch to zeroconf

stephanemagnenat, 2017-11-17:
There are two problems actually:
1. the switch is not advertised on zeroconf (see #697)
2. the e-puck do not stop advertising when connected

stephanemagnenat, 2017-11-17:
Problem 2 is due to a race condition when a target that has not been registered yet is already removed. Zeroconf requires some protection against this.

stephanemagnenat, 2017-11-17:
It seems that removing the targets after registration is complete is not sufficient to prevent the race condition. This requires more investigation.

stephanemagnenat, 2017-11-17:
Commit bc18326 solves the race condition by keeping a set of targets that are requested for deletion before being fully registered, and deleting them once their registration is completed.

commit d5c6dd86 (2017-11-20): Fixed #719
closed 2017-11-20 by stephanemagnenat