aseba-community/aseba #724: filter target in connection windows

mbonani, 2017-11-29:
often lot of  target are displayed  that are not relevant (BT com port, serial port on Linux). Also with zeroconf and simulator you can have a big list.
I know that we want to keep it as general as possible but could we imagine check boxes to set up a filter like:
"show all", "show only local", "show Thymio" "show simulated robot" "show network robot" "do not show unknow device" etc.
 to decide what is generic or not, check box can perhap be exclusif or not,

stephanemagnenat, 2017-11-29:
In the long run is a good idea. Maybe the Thymio connection dialogue could be a specialisation of this box.

davidjsherman, 2017-12-01:
As in #727, we should decide what descriptive data is applied to each node, and standardize this with the Zeroconf description. Considering the examples,

* show Thymio: *type* is "Thymio"
* show simulated: *type* is "Playground"
* show network: underlying stream is "tcp:"? Should we exclude those that resolve to *localhost*?
* show only local: underlying stream is not "tcp:"? Should we include *localhost*?

It seems unfortunate that we would need to interrogate the Dashel stream.

Do we want the standard description to include arbitrary tags? They could be added to the TXT record.