aseba-community/aseba #725: Timer in VPL should be stoppable

mbonani, 2017-12-01:
I discover that I cannot stop timer in VPL, I was expected if I put zero it will stop it. Instead it is put at 1. Here the code I wanted to do:
[image]

I know that I can find a way wis another state, but it was coherent for me to stop timers.

riedo, 2017-12-01:
but here the problem is that you think of it as a timer but it is just a delayed event and you choose the time to trigger it. I saw people getting confused when moving on to text programming because of that.

Basically the philosophy is that:
**VPL:** The timer action triggers a _one-time_ event with a delay. If you put  the time at 0 it will be triggered right away. I never saw people trying to stop the timer by putting 0
**Text:** You can set the period of a timer that will happen _regularly_ until you stop it. you stop it by setting the period to 0 (and because he was used to that, Michael thought to do the same in VPL). I saw people trying to to in text programming as they would do in VPL, i.e. when they want to trigger an action with a delay the write timer.period[0]=4000 and then expect that from the moment this line is executed, 4 seconds later the onevent timer0 code will be executed once.

davidjsherman, 2017-12-01:
Fanny, if I understand you, there are two different things.

In VPL, you have a **timer** (*minuteur*), which will always trigger an event after a (possibly zero) delay. The event will always happen; to cancel it you can trigger a state change, with rules <sup>:warning:</sup> guarded by that state that absorb the timer event. 

In Aseba, you have a **clock** or an **oscillator**, whose period you can set. Curiously, to cancel the oscillator one sets its period to 0, whereas logically one would expect that to mean that the event fires continuously without delay.

Confusion stems from storing the activation of the clock *in band* in the value of the period, whereas it should be out of band. In Aseba maybe one can use hand-waving and say, *positive values define a period, non-positive values inactivate the clock*, but something like this in VPL is asking for trouble.

Furthermore, it is plausible that one would to want to trigger a VPL timer event immediately: for example it might be an easy way to jump to the start state of the automaton.

So I think that the reasonable answer is a new VPL action that cancels the timer:
[image]

:warning: You potentially need many rules because you have to reset the state to the one you had when you wanted to cancel the timer, in fact I think you need as many absorbing states as original states. This is pretty ugly!