aseba-community/aseba #727: give custom names to tabs

riedo, 2017-12-01:
It could be convenient to give custom names to tabs or to windows when working on several robots at the same time to find which is which.
(for example If I use a simulated R2T2 playground and have several Studio connected to different Thymio, it starts to be confusing)

davidjsherman, 2017-12-01:
We would want this information associated with the target and made available using Zeroconf. As with #724, we need to define what node description data needs to be associated with every node.

Currently we associate the following information with each node:
* **name**: human-readable node name
* **type**: human-readable node type
* **ids**: Aseba node ids, there may be several if the target is a switch
* **pids**: Aseba product ids, there may be several if the target is a switch
* **busy**: flag to say the node is associated with a client

The **name** is the Zeroconf mDNS name, the other four are stored in the TXT record.

The name can be freely modified. I think that the Studio tab label and the node name should be the same.

cor3ntin, 2017-12-06:
Another idea would be to set a name for each thymio on the robot memory, complementing the pid.
The name could be changed from the connection window.

davidjsherman, 2017-12-06:
I think that the proper place to put the name, is in the entity that will be handling Zeroconf responses.  For the Thymio IIs this is the Switch (although it could be the Studio if some day we add web bridge to Studio).

A dedicated Switch in a classroom could remember an id--name mapping, but honestly I expect that students will be renaming the robots all of the time, so forgetting the robot names could actually be seen as a feature.

The Thymio Wifi would presumably handle Zeroconf itself, and would have memory available to store the name. If that were desirable.