aseba-community/aseba #729: Onevent buttons called when program starts

Thyrf, 2017-12-06:
I would like to create a program which will do something when the center button in pressed and put this program on an SD card. The problem is that when I put such a program on an SD card,put it in the robot ,start the robot and press the center button to start my program, the program is not only initialised but the event depending on the center button of my program is launched as well.

For instance if I put the program corresponding to this code : 

onevent buttons
	when  button.center == 1 do
		call leds.top(32, 0, 0)
	end

and either write it on the robot or put it through the SD card, then when I select the 7th mode corresponding to my program, the event is triggered and the robot turns red as soon as the program starts... While I would expect it to happen only when I press the button once my program has started.

Thyrf, 2017-12-06:
Yes indeed I am using a physical Thymio.
My firmware version is the 10.
Should I post this on aseba-target-thymio2 then ?

stephanemagnenat, 2017-12-06:
@mbonani did you already encounter this problem?

marvelous, 2017-12-07:
This behavior is on purpose: when the program starts, true conditions trigger events immediately. Later, they trigger only on a rising edge.

The opposite behavior brings issues such as https://github.com/aseba-community/thymio-vpl2/issues/8#issue-194081872 and https://github.com/aseba-community/thymio-vpl2/issues/23#issue-194930450. The VPL2 implementation doesn't use the `when`s, so we had to retrofit this program startup special case behavior in VPL2 to fix the above issues.

Thyrf, 2017-12-07:
Then I might not have understood the "when" correctly. I though when button.center == 1 is true only if button.center was not 1 before. Isn't that so ?

And if it is so then button.center is equal to 1 when I start my program so that when my when is called then as my condition of having button.center equal to a different value than 1 not respected, I would expect the condition to be False. Wouldn't that make sense ?

marvelous, 2017-12-07:
> Then I might not have understood the "when" correctly. I though when button.center == 1 is true only if button.center was not 1 before. Isn't that so ?

It is, except when the program starts, because the `when`s are considered to have been checked as false before the program starts. Under the covers, there is a bit for every `when` in the script, and all these bits are initialized to 0 when the program starts.

Unfortunately, the `when` implementation has this edge case and now people rely on it: they want their line following program to react as soon as the program starts.

Now I totally agree that the behavior you want makes more sense, it's what I had implemented in VPL2 initially. I am only exposing why it works this way and why it's not trivial to change it. Any idea on how to reconcile both cases is welcome!

Thyrf, 2017-12-07:
Ah, that's interesting. I happened to wonder why the line following was forging the way it does on starting. Ok I get it, and I understand your decision of implementation, thank you.

stephanemagnenat, 2017-12-07:
@Thyrf should we close the issue or you wish to keep it open for further discussion?

Thyrf, 2017-12-07:
We can close it. Thank you.

closed 2017-12-07 by Thyrf