aseba-community/aseba #740: Update translations

stephanemagnenat, 2017-12-08:
Update translations for Studio, VPL, Thymio firmware updater and Thymio wireless configurator.
- [x] French translation
- [ ] German translation
- [ ] Italian translation
- [ ] Spanish translation
- [ ] Greek translation
- [ ] Chinese translation

basilstotz, 2017-12-08:
If you need a hand, I could make the german translations. Any instructions?

stephanemagnenat, 2017-12-08:
Thanks a lot! Yes, you can the files ending in `_de` from [here](https://github.com/aseba-community/aseba/wiki/translation-release-1.6), and translate them using `qlinguist` as explained [here](http://doc.qt.io/qt-4.8/linguist-translators.html), the general procedure being explained in our [localization guide](https://github.com/aseba-community/aseba/blob/master/localization.md). If you are uncomfortable with git and pull requests, you can send me the updated files by email.

paraskevasleivadaros, 2017-12-11:
Hello, I'm a native Greek speaker and I would be more than happy to make the Greek translations. Let me know if I can help!

stephanemagnenat, 2017-12-11:
Thank you very much! The same instruction apply for Greek as German, but with the `_el` postfix in the names.

mbonani, 2017-12-12:
Paolo will work on italian translation.
You can find standalone qt linguist installer here. https://github.com/lelegard/qtlinguist-installers/releases

stephanemagnenat, 2017-12-18:
We are now only missing Chinese and Greek translations, and none of them are release blockers, so I mark this item as done in #596 and keep it here for further bookkeeping.

stephanemagnenat, 2018-01-09:
The Chinese translation is in, any news from the Greek one?

paraskevasleivadaros, 2018-01-11:
Sorry about that, i had a lot of work and i forgot about them. I will try and make them until the end of this month. I hope that's okay. If not, let me know and i will do my best to have them delivered as soon as possible! :)

paraskevasleivadaros, 2018-02-11:
Hello, i wanted to inform you that today or tomorrow the translations will be ready!

mbonani, 2018-03-07:
greek is done

closed 2018-03-07 by mbonani