aseba-community/aseba #746: caractarer like é ü are not accept to compile in macosx

mbonani, 2017-12-12:
aseba on windows (probably linux)  accept in the code accentuate character. 
on macosx it make error " indentifier must begin with _ or an alpha numeric character, found unicode character instead"
In the description of the language it is normally not permit. But generated code coming from blockly could have some. We should be consistent.

stephanemagnenat, 2017-12-13:
On my mac is works.

stephanemagnenat, 2017-12-14:
It seems some locales do not consider unicode characters to be alphanumeric. According to [cpp reference.com](http://en.cppreference.com/w/cpp/locale/isalpha), it could be forced. Maybe forcing the locale to "UTF-8" could help.

stephanemagnenat, 2018-01-09:
Do we move that to 1.6.1?

mbonani, 2018-01-09:
yes

ypiguet-epfl, 2018-02-23:
The error message has typos which make it difficult to find in the source code. It's _Identifiers must begin with _ or an alphanumeric character, found unicode character 0x?? instead_ where _0x??_ stands for the hexadecimal unicode code point for the Basic Multilingual Plane (up to U+FFFF), something else beyond. It's for error `ERROR_INVALID_IDENTIFIER` thrown by method `Aseba::Compiler::tokenize` in [aseba/compiler/lexer.cpp](https://github.com/aseba-community/aseba/blob/master/compiler/lexer.cpp).

Suggestion: write your own unicode handling code and don't depend on Qt, c++ standard libraries, the OS, user settings, or any short-term hack.

cor3ntin, 2018-02-23:
The underlying issue have been found, it's indeed because the tool rely on the underlying environment settings ( which it should not) and for some reason on OSX that environment may not be utf8.
Expect a fix in the coming weeks.

However, we should definitively not reinvent the wheel. Why locale support in the STL is somewhat lacking, it's top of the class in Qt, when used properly.

ypiguet-epfl, 2018-02-23:
What about tools which use the compiler without a GUI, such as the switch? In mine, when compiled on macOS, I get the same error:
```
./vmshell --dis --src -
var é
Error at Line: 1 Col: 5 : Identifiers must begin with _ or an alphanumeric character, found unicode character 0xe9 instead
Compilation error.
```
Unicode character categories and utf-8 aren't rocket science. A platform-independent, Qt-independent, definitive solution would be nice.

cor3ntin, 2018-02-23:
Unicode is anything but simple, especially if we don't want a dependency on icu ( which Qt has on most systems).
For this particular issue It should be doable to have a reasonably well behaving solution, don't worry.

ypiguet-epfl, 2018-02-24:
While the whole unicode standard is large and [prone to bugs](https://techcrunch.com/2018/02/15/iphone-text-bomb-ios-mac-crash-apple/), that isn't needed by the compiler or any Aseba non-graphical tool. Classifying code points between what's valid as first character in an identifier, what's valid as remaining characters in an identifier, and everything else would be enough. If by "unicode is anything but simple" you mean the text conversion to canonical forms, it's uncommon: [Javascript explicitly doesn't do it for identifier matching](http://www.ecma-international.org/ecma-262/6.0/#sec-names-and-keywords), for instance. I'm not impressed by Qt unicode support, which fails to display reliably "e acute" with a combining acute accent (U+0065 U+0301) in Studio 1.6 on Mac.

I don't worry, I just hope there won't be dependencies difficult to fulfill or huge for the simple tools.

cor3ntin, 2018-02-24:
See https://github.com/aseba-community/aseba/pull/842/commits/09835e6788ab3dd518df7b1088afacabd7a3b421#diff-0e1c4b3b848444324ad5dc75a312633cR195 for a temporary fix and a detailed explanation.

Nothing is simple if you want to do things properly, and I'm not a huge fan of half-backed, wheel reinventing solutions.

cor3ntin, 2018-02-28:
@stephanemagnenat what's your take on that ?

cor3ntin, 2018-03-07:
It has been merge in master.
A PR for 1.6 has been made too https://github.com/aseba-community/aseba/pull/849

cor3ntin, 2018-03-07:
Fixed

closed 2018-03-07 by cor3ntin