aseba-community/aseba #752: Subroutine calls from subroutines don't work

ptpj, 2017-12-16:
In blocky, an attempt to call a subroutine from another subroutine fails: the calling procedure has empty body in the generated VPL code. Calls from event handlers work as expected.

[image]

stephanemagnenat, 2017-12-17:
This looks like a bug for [aseba-community/blockly](https://github.com/aseba-community/blockly), where the Blockly to Aseba compiler lies.

ptpj, 2017-12-17:
Looks like this is intentional / known caveat:

https://github.com/aseba-community/blockly/blob/866c66c66df78100816c77d5bd8879edacf06927/generators/aesl/procedures.js#L18

stephanemagnenat, 2017-12-17:
Ok, then it should be documented. @mbonani could you please take care of that?

ptpj, 2017-12-17:
It actually seems the statement is not true anymore, is it? I seem to be able to define subroutines in AESL in arbitrary order regardless of how they call each other (as long as I don't introduce recursion).

So perhaps the if block can be removed altogether?

davidjsherman, 2017-12-18:
I don't think it is appropriate to remove the subroutine block. Everything expressible in Thymio AESL should be expressible in Thymio Blockly.

It is possible to teach Blockly to disallow certain kinds of connection, such as nested subroutine blocks. Could you move this issue to [aseba-community/blockly](https://github.com/aseba-community/blockly) so that I can document the mechanisms there?

ptpj, 2017-12-19:
Moved to https://github.com/aseba-community/blockly/issues/21

closed 2017-12-19 by ptpj