aseba-community/aseba #757: Community communication tool

stephanemagnenat, 2017-12-19:
In the past we were using mailing lists on gna, but they shut down some months ago.

@Cor3ntin has suggested to set up a new communication tool, he mentioned [riot.im](https://riot.im) or [Slack](https://slack.com/). We could also set-up again mailing lists, Eigen for example uses [tuxfamily](http://eigen.tuxfamily.org).

I think that community members should express their needs and then we'll be in good position to select the right tool.

stephanemagnenat, 2017-12-19:
One thing I miss since the shut down of the mailing lists is the lack of a clear place to have generic discussions related to Aseba. An obvious solution to this problem would be to create new mailing lists.

Vincebecker, 2017-12-19:
I am not sure it's the best solution since I am not the real "user" here if you talk about developer community.

But, please, have a look at mobilize.io it's sort of a forum and you can also answer and discuss using mail if you don't want to use the interface.

stephanemagnenat, 2017-12-19:
A forum is also an interesting option, I will look at mobilize.io.

cor3ntin, 2017-12-19:
Another solution would be https://about.mattermost.com . An open source slack clone.

Ideally
 * Free ( as in free beer)
 * Low/no maintenance as we don't really have IT resources to spare ( which is my concern with riot.im for example). Slack works "out of the box"
 * Supports both offline/async mode & online mode with persistence
 * Voice/video support so Mobsya can get rid of 1-1 messaging software such as skype
 * Easy to use for non devs
 * Same tool for mobsya / the wider community as to avoid multiplying the number of tool used, but maybe the needs are different.

cor3ntin, 2017-12-19:
An alternative forum option could be https://www.discourse.org/ (open source )
Discourse is more a forum, while mobilize.io is closer to google group

marvelous, 2017-12-19:
@cor3ntin has given 3 instant messaging examples, but @stephanemagnenat talks about mailing list and @Vincebecker about forums. In my opinion, forums and mailing lists have almost the same use, but instant messaging is a very different beast! Who will use that tool? What for?

Other options:

- github is fine for general discussion, like in aseba-community/thymio-vpl2#116
- http://gitter.im/ has great github integration
- the Thymio forums

But we should mostly talk about use cases.

cor3ntin, 2017-12-19:
@marvelous I totally agree.

I think there are two use cases, and they are quite different

-  A tool for the community to share content, discuss about the project and help each other out, as well as maybe get news, etc. That's what @Vincebecker is talking about and he knows more about that than me. But a forum seem suitable

- An IM tool for the people working on aseba and related tools (including non dev) to easily reach out and have group discussions, etc. That would replace skype/hangout/whatever. But maybe the community could have also access to that IM - maybe on a different sets of room, I don't know.

stephanemagnenat, 2017-12-19:
Indeed these two usage are very different. We can go a bit more in detail. I see these different use cases:
- A new contributor wants to contact or discuss something with the developers. For me the most natural way is a mailing list. A forum could do if there is an email bridge, as a solution that forces to pull content manually would not work for busy people.
- A user has a question or wants to share something. A forum seems the most natural solution. The question for Aseba is how many Aseba users who are not Thymio users want to use such a tool. If there are several, we should have an Aseba forum, but until then, I'm not very sure it's helpful because on the one hand it might confuse Thymio users. But of course on the other hand it could also help to build an Aseba-specific user community and maybe help clarify the confusion between Aseba and Thymio.
- Private, non-archived real-time discussion between two or more Aseba developers. Currently we use a mix of phone, Skype, appears.in, hangout and physical meetings. We can probably improve that, but the problem is that most of us have different projects at the same time so we might still use the communication mechanism that is right under our hand.
- Public, archived real-time discussion between two or more Aseba developers. We do not do that currently. Traditionally open-source projects use IRC for that. I think that this category applies only for chat, and I am not sure that it is very necessary, because talkincg while updating github issuos work fine. But maybe other people feel the need for that.