aseba-community/aseba #759: Style cleaning

cor3ntin, 2017-12-19:
Both `dashel` and `aseba` have some code styling inconsistencies that makes t difficult to not introduce some ws-only change or styling issue.

Formatting can hopefully be automated away which leads to increased productivity in the long run.

I've made some experiment on dashel to that aim https://github.com/asebacommunity/dashel/compare/master...cor3ntin:cleanws

-  Convert tab to space
`find \( -name "*.cpp" -o -name "*.h" \) -exec bash -x -c 'unexpand -t 4 "$0" | sponge "$0"' {}  \;`
- Remove trailing ws
`find \( -name "*.cpp" -o -name "*.h" \) -exec bash -x -c 'sed -i"s/[ \t]*$//" "$0"  ' {}  \;

-Provide a custom clang format style that matches closely the style of `aseba`. That can be improved further.
https://github.com/cor3ntin/dashel/blob/eacaa85695ea433ab12ee800d5845530e0afd20b/.clang-format

Then it becomes very to ensure a consistent style.
The issue of course is that it introduces a big formatting change across of which it is painful to merge.
Timing the introduction of the change properly helps, otherwise conflict will have to be solve manually,
which I think is a small price to pay going forward.

Or we could only enforce the use of the tool on new files and diffs

https://clang.llvm.org/docs/ClangFormatStyleOptions.html
https://github.com/llvm-mirror/clang/blob/master/tools/clang-format/git-clang-format
https://dx13.co.uk/articles/2015/4/3/Setting-up-git-clang-format.html

stephanemagnenat, 2017-12-20:
Please do not force space, we have been using tabs for years because we like them.

For the rest yes it's good to enforce a consistent style.

stephanemagnenat, 2017-12-20:
Also regarding maintaining identation on newlines: I am not convinced it is bad, it rather feels to me that the creator of git disliked it and decided to put it in red to force people into their convention. But if one gives more arguments on why it is better, I'm ready to neconsider my decision.

stephanemagnenat, 2017-12-20:
Note that this gives me a 404: https://github.com/asebacommunity/dashel/compare/master...cor3ntin:cleanws, probably it is private.

cor3ntin, 2017-12-20:
Sorry, I meant space to tab.

cor3ntin, 2017-12-20:
`https://github.com/aseba-community/dashel/compare/master...cor3ntin:cpleanws`

stephanemagnenat, 2017-12-20:
> Correct url is https://github.com/aseba-community/dashel/compare/master...cor3ntin:cpleanws

This page gives:
There isn’t anything to compare.
We couldn’t figure out how to compare these references, do they point to valid commits?

cor3ntin, 2017-12-20:
> Correct url is aseba-community/dashel@master...cor3ntin:cpleanws

Github keeps messing with the url `
https://github.com/aseba-community/dashel/compare/master...cor3ntin:cleanws`
- I wanted to avoid making a PR yet.