aseba-community/aseba #760: asebamassloader is not stable on windows in playground

mbonani, 2017-12-19:
I notice instability with asebamassloader on my computer with the simulator. I test with latest build and yeti discovery kits. 
Few times it works message on simulator consol are correct and behavior is charged

[image]

Some time message are correct (perhaps order is wrong) and behavior is not launched:

[image]

Most time message are showing problems: 
[image]

There is probably a timing issue

davidjsherman, 2017-12-20:
Does it work if you  replace `asebamassloader` by `asebahttp` in the playground file, and hide the robot behind the `asebahttp` switch? (The switch would run permanently on port 33333, with only one robot node in its network on, say, port 33334.)

Because with `asebahttp` one is guaranteed that the AESL program is fully compiled and loaded, before the switch functionality is started.

stephanemagnenat, 2017-12-20:
I agree that it is likely a timing issue. @davidjsherman's approach would work but require running http yor the duration of the session (which could be a good think from a blockly/scratch perspective).

Another approach could be to add some delays to the process option to let it launch later.

I guess that for 1.6, the http approach is probably easier.

stephanemagnenat, 2017-12-20:
Looking at the error in the last screenshot and at the medula code (where there is a one second delay after a successful connection until the code is uploaded), I am not so sure that it is a delay problem related to the description of the target. On the faulty screenshot, the code seems indeed to be loaded properly.

stephanemagnenat, 2017-12-20:
After some thought, I believe that the problem is that massloader would just close after the Run() message had been sent and the stream flushed, but on Win32, stream::flush() does not ensure a flush. I added another 1s second of wait in 0d2927bf, hopefully it solves the problem.

mbonani, 2017-12-20:
I will test,
but yes it is working with asebahttp, it is also interesting to have the simulated Thymio ready with web bridge. It could simplify for the user. 

Also should we load each time the default behavior on all Thymio (in the R2T2 world)?
Only on some world? in that case should we have a naming convention at the file to know? Should we wait planned improvement of 1.7?

davidjsherman, 2017-12-20:
See PR #761

cor3ntin, 2017-12-20:
I don't think starting an asebahttp from the playground is a good idea.
The issue being, it can't be discovered with a non default port, since there is no web API to do service discovery, so you still need some kind of process to handle that, sadly.
Worse, the port could already be in use.

Chrome has an *extension* API for service discovery, maybe the same thing could be implemented for Firefox, but it still does not change the fact that we need something between the web page and the device in order to make a connection.

However, launching an asebahttp could register the current target in an already running asebahttp instance, meaning that there would only ever be a single asebahttp instance running, on a fixed port.

davidjsherman, 2017-12-20:
Which port are you talking about?
* The http port should be chosen from the ephemeral range and advertised using Zeroconf
* The public aseba port (33333) is the same one originally advertised by the playground
* The internal port (33303) is from the unassigned range

If you want the communication between Http and Playground to be truly out of band, then Playground could offer its target as a local socket connection. Http through Dashel can use these connections.

cor3ntin, 2017-12-20:
The http port can't be advertised using zeroconf, because the local webpage can't discover it. So there needs to be a single http port per host.
The rest can be on an ephemeral ports indeed.

davidjsherman, 2017-12-20:
For #761 commit d11ffe5d4656ffc2ffebd86bbbf3bcbd5fe694aa specifies an ephemeral HTTP port to avoid conflicts on port 3000

cor3ntin, 2017-12-20:
yep, and that doesn't work ( because, blockly assumes a fixed port and we can't work around that )

davidjsherman, 2017-12-20:
Re: https://github.com/aseba-community/aseba/issues/760#issuecomment-353022082, why can't the web page use e.g. `bonjour.find` from [npm package bonjour](https://www.npmjs.com/package/bonjour)?

cor3ntin, 2017-12-20:
> Re: #760 (comment)

The npm package works with node, not from a web page.
Web sockets are not actual sockets. We'd need a browser extension.

There was a proposal https://www.w3.org/TR/discovery-api/ -  but it's dead
in the water.

davidjsherman, 2017-12-20:
Also, I fear that this is confusing the issue. The desired behavior is to run Playground with a Studio or VPL connection to a discoverable Aseba TCP target. PR #761 loads the AESL behavior in the Thymio Discovery examples using `asebahttp`, in order to guarantee that the file loads. As an added benefit, running the HTTP switch makes it possible to connect a Blockly or ScratchX client.

* If we want the added benefit, then currently we need to use the standard Aseba HTTP port. In that case, cherry-pick 2c4386cdc97331ed2b88ab8875e929e23483e3ca.
* If not, in order to avoid a conflict with an Aseba HTTP that is already running, we need to tell the switch to use an ephemeral port. In that case, use the head commit d11ffe5d4656ffc2ffebd86bbbf3bcbd5fe694aa.

davidjsherman, 2017-12-20:
@mbonani those are some good ideas in https://github.com/aseba-community/aseba/issues/760#issuecomment-352996423. Always loading the default behavior would make the simulator seem more like the real robot, and make it easier for teachers to train at home.

It is easy to add it systematically to all of the playground files. No software modifications are necessary because Playground can't start without a playground file. Proposed in PR #762.

The R2T2 worlds are a case where Massloader is a better approach, because we want all of the 4 or 16 Thymios to be individually addressable. Using Http would join them into a single network, controlled by a single client.

davidjsherman, 2017-12-20:
@cor3ntin https://github.com/aseba-community/aseba/issues/760#issuecomment-353022082 true, service discovery is properly the responsibility of the switch. I just remembered #721, which allows the switch to act as a broker.

cor3ntin, 2017-12-20:
@davidjsherman shouldn't it be the other way around ?
The switch should probably auto-connect to all local targets ( aka - on the same compuler ), and then have a setting to decide whether these targets should be exposed on the network ?

davidjsherman, 2017-12-20:
Moved to #764 since this discussion isn't germane to #760

stephanemagnenat, 2017-12-20:
@mbonani and others, did my change 0d2927b fixed this issue? If so, please close it.

mbonani, 2017-12-20:
yes perfectly

closed 2017-12-20 by mbonani