aseba-community/aseba #767: Compilation error: Polygone does not name a type

albinou, 2017-12-25:
Hi and thanks for this nice project :)

I tried to compile the branch _release-1.5.x_ of aseba. I followed the instructions on this [page](/aseba-community/aseba/blob/master/compile.Linux.md) and I had a compilation issue with the _Polygone_ type. Renaming the type to _Polygon_ fixed it.

Would it be desirable to apply the commit 97e99992bb5ac8ad5d9c625c891a455487514ee2 (of master) into the _release-1.5.x_ branch?

Thanks.

Reference to Enki type update: [issue #42](/enki-community/enki/issues/42)

davidjsherman, 2018-01-02:
Hi, thanks for reporting this issue. I am not sure that back porting Enki for Aseba 1.5.x is feasible, but I would expect that compiling [Aseba branch 1.5.x](https://github.com/aseba-community/aseba/tree/release-1.5.x) using the [release of Enki tagged "aseba-1.5.4"](https://github.com/enki-community/enki/releases/tag/aseba-1.5.4) would work.

I've added the "Documentation" tag since the instructions in [compile.Linux.md](https://github.com/aseba-community/aseba/blob/master/compile.Linux.md) mix the release 1.5.4 branch of Aseba with the master branch of Enki, rather than specifying the appropriate release.

albinou, 2018-01-02:
Oh thanks for your comment.
I had not seen there was a tag for version 1.5.4 in the Enki project.

closed 2018-04-17 by mbonani