aseba-community/aseba #770: firmware version of simulated Thymio

mbonani, 2018-01-04:
I will release soon the firmware 11 of Thymio and it will implies several question:

- should we update the firmware version also on simulated robot?

Note: we should also be careful with .aesl files that are given with the package to be consistent

mbonani, 2018-01-04:
.aesl are not a problem... I add a strange thing that confuse me

davidjsherman, 2018-01-09:
This issue is still open but it seems that @stephanemagnenat made the change in 4bf4b8c8e95e045b125f7423e88b710f21d5d6aa.

I think yes, the firmware version should be the same. Especially if the simulator with the aesl files now has all of the behaviors of the true firmware.

At the risk of [bikeshedding](https://fr.wikipedia.org/wiki/Loi_de_futilité_de_Parkinson), is the firmware version determined only by the VM code in Aseba, or by the code in the firmware target? Because in an ideal world we would define the version in only one place.

Since aseba-community/aseba-target-thymio2#29 pulls in `aseba/vm/vm.h`, it is tempting to define the version in Aseba and propagate it to [aseba-target-thymio2/main.c](https://github.com/aseba-community/aseba-target-thymio2/blob/master/main.c#L41) and [aseba/targets/playground/ robots/thymio2/Thymio2.cpp](https://github.com/aseba-community/aseba/blob/master/targets/playground/robots/thymio2/Thymio2.cpp#L60). But I guess this is backwards: it should be possible to bump the true firmware version even if the VM definition stays the same.

stephanemagnenat, 2018-01-09:
> Since aseba-community/aseba-target-thymio2#29 pulls in aseba/vm/vm.h, it is tempting to define the version in Aseba and propagate it to aseba-target-thymio2/main.c and aseba/targets/playground/ [...]

I think that the proper approach is to define for every target an API version, which might or might not increase along with the firmware version. These API could be described in the Aseba documentation and implemented by several targets, real or virtual. The couple (API version, product Id) would uniquely define feature sets from the point of view of client code (VPL, Scratch, Blockly, etc.). I would also considering moving productId and fwversion from mandatory hidden variables into the target description itself (that would bump protocol version). They could still be present as hidden variables for the sake of backward compatibility.

closed 2018-01-09 by stephanemagnenat