aseba-community/aseba #776: Keyboard and mouse actions in asebachallenge described in help not working

ypiguet-epfl, 2018-01-09:
Some of the shortcuts described in the help message in `asebachallenge` don't work (tested on Ubuntu 16.10). E.g. F1 for displaying the message again, or mouse wheel.

stephanemagnenat, 2018-01-09:
That is a bug indeed, challenge does not suppress the help icon as it should, so it is showing the help of Enki playground instead.

stephanemagnenat, 2018-01-09:
Fixed by ad87974 in Aseba and enki-community/enki@a51981112f1a26b5b83c55bc193ba8333574e60b

closed 2018-01-09 by stephanemagnenat

davidjsherman, 2018-01-11:
@stephanemagnenat this change fails in CI on macOS and Windows ([console log](https://ci.inria.fr/inirobot/job/aseba-community/job/aseba/job/master/34/console)):
```
[Compile on macos] /Volumes/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q4UUWZZOLNUTQURMBB2XXA/targets/challenge/challenge.h:73:25: error: only virtual member functions can be marked 'override'
[Compile on macos]                 void displayWidgets() override;
[Compile on macos]                                       ^~~~~~~~
[Compile on macos] /Volumes/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q4UUWZZOLNUTQURMBB2XXA/targets/challenge/challenge.h:74:40: error: only virtual member functions can be marked 'override'
[Compile on macos]                 void clickWidget(QMouseEvent *event) override;
[Compile on macos]                                                      ^~~~~~~~
[Compile on macos] /Volumes/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q4UUWZZOLNUTQURMBB2XXA/targets/challenge/challenge.h:80:8: warning: 'helpActivated' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
[Compile on macos]                 void helpActivated();
[Compile on macos]                      ^
[Compile on macos] /Volumes/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q4UUWZZOLNUTQURMBB2XXA/dist/macos/include/viewer/Viewer.h:253:16: note: overridden virtual function is here
[Compile on macos]                 virtual void helpActivated();
[Compile on macos]                              ^
[Compile on macos] 1 warning and 2 errors generated.
[Compile on macos] make[2]: *** [targets/challenge/CMakeFiles/asebachallenge.dir/challenge.cpp.o] Error 1
[Compile on macos] make[1]: *** [targets/challenge/CMakeFiles/asebachallenge.dir/all] Error 2
[Compile on macos] make: *** [all] Error 2
```
```
[Compile on windows] C:/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q
[Compile on windows] 4UUWZZOLNUTQURMBB2XXA@2/targets/challenge/challenge.h:73:8: error: 'void 
[Compile on windows] Enki::ChallengeViewer::displayWidgets()' marked 'override', but does not 
[Compile on windows] override
[Compile on windows]    void displayWidgets() override;
[Compile on windows]         ^~~~~~~~~~~~~~
[Compile on windows] 
[Compile on windows] C:/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q
[Compile on windows] 4UUWZZOLNUTQURMBB2XXA@2/targets/challenge/challenge.h:74:8: error: 'void 
[Compile on windows] Enki::ChallengeViewer::clickWidget(QMouseEvent*)' marked 'override', but does 
[Compile on windows] not override
[Compile on windows]    void clickWidget(QMouseEvent *event) override;
[Compile on windows]         ^~~~~~~~~~~
[Compile on windows] 
[Compile on windows] make[2]: *** [targets/challenge/CMakeFiles/asebachallenge.dir/build.make:89: 
[Compile on windows] targets/challenge/CMakeFiles/asebachallenge.dir/challenge.cpp.obj] Error 1
[Compile on windows] 
[Compile on windows] make[1]: *** [CMakeFiles/Makefile2:1893: 
[Compile on windows] targets/challenge/CMakeFiles/asebachallenge.dir/all] Error 2
[Compile on windows] 
[Compile on windows] make: *** [Makefile:141: all] Error 2
```

davidjsherman, 2018-01-11:
@stephanemagnenat My mistake, the upstream build of Enki wasn't triggered. See enki-community/enki#48