aseba-community/aseba #778: Ambiguous call of make_unique in tree-expand.cpp in c++14

ypiguet-epfl, 2018-01-15:
When compiling tree-expand.cpp with c++14 or c++17, gcc fails. No error with c++11.
Command:
```
g++ -std=c++14 -g   -c -o tree-expand.o compiler/tree-expand.cpp
```
Output:
```
compiler/tree-expand.cpp: In member function ‘virtual Aseba::Node* Aseba::MemoryVectorNode::expandVectorialNodes(std::wostream*, Aseba::Compiler*, unsigned int)’:
compiler/tree-expand.cpp:317:83: error: call of overloaded ‘make_unique(Aseba::SourcePos&, unsigned int&, unsigned int&, std::__cxx11::wstring&)’ is ambiguous
 rray = make_unique<ArrayWriteNode>(sourcePos, arrayAddr, arraySize, arrayName);
                                                                              ^
In file included from compiler/tree-expand.cpp:23:0:
compiler/../common/utils/utils.h:74:21: note: candidate: std::unique_ptr<_Tp> Aseba::make_unique(Args&& ...) [with T = Aseba::ArrayWriteNode; Args = {Aseba::SourcePos&, unsigned int&, unsigned int&, std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&}]
  std::unique_ptr<T> make_unique(Args&&... args)
                     ^
In file included from /usr/include/c++/5/memory:81:0,
                 from compiler/../common/utils/utils.h:30,
                 from compiler/tree-expand.cpp:23:
/usr/include/c++/5/bits/unique_ptr.h:764:5: note: candidate: typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = Aseba::ArrayWriteNode; _Args = {Aseba::SourcePos&, unsigned int&, unsigned int&, std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<Aseba::ArrayWriteNode, std::default_delete<Aseba::ArrayWriteNode> >]
     make_unique(_Args&&... __args)
```
Suggestion: in [tree-expand.cpp](https://github.com/aseba-community/aseba/blob/9802ba911abbc491f7741fca6e808ea32353e7bf/compiler/tree-expand.cpp), lines 317 and 319 in method `MemoryVectorNode::expandVectorialNodes`, prefix `make_unique` with `Aseba::` to have
```
  if (write == true)
    array = Aseba::make_unique<ArrayWriteNode>(sourcePos, arrayAddr, arraySize, arrayName);
  else
    array = Aseba::make_unique<ArrayReadNode>(sourcePos, arrayAddr, arraySize, arrayName);
```

stephanemagnenat, 2018-01-15:
Which version of g++ are you using? The command you give compiles fine with g++ 5.4.0 on Ubuntu 16.04.3.

stephanemagnenat, 2018-01-15:
I removed the `make_unique()` template from Aseba in commit 1a2974d306ee90656e860f6e263535c411e0b950.

ypiguet-epfl, 2018-01-15:
Sorry, big error on my part. It was an old commit. Everything is fine in the current version with `std::`.

closed 2018-01-15 by ypiguet-epfl

cor3ntin, 2018-01-15:
@stephanemagnenat You should **never** put anything in `std::` ( beside customization points ) -  we should remove reference wrapper too.

stephanemagnenat, 2018-01-15:
@cor3ntin I disagree, this was the official suggested fix by Scott Meyers in "Effective Modern C++" to the lack of support of `make_unique` in C++11.

cor3ntin, 2018-01-15:
It's fine. Just not in `std`
http://eel.is/c++draft/namespace.constraints#namespace.std-1

````
namespace polyfill {
#if HAS_MAKE_UNIQUE
     template <typename...T>
     using make_unique = std::make_unique<T....>;
#else
      template <typename...T>
      std::unique_ptr<T> make_unique(T &&....) {}; 
#endif
}
```

cor3ntin, 2018-01-15:
Not that in this particular case it's not undefined, it's ill-formed. make_unique is already defined so it's an ODR violation.