aseba-community/aseba #788: Can't find abi:cxx11 tagged method from Dashel dynamic library on Windows

davidjsherman, 2018-01-17:
Aseba should be able to compile with the default build of Dashel. By default, Dashel builds with shared libraries. Unfortunately, Aseba will only compile with a static library version of Dashel on Windows.

Specifically, Aseba::Zeroconf asks Dashel to return a target parameter as a string. Template handling must have changed in C++11, because it is an ABI C++11 version of the method that can't be found:
```
Linking CXX executable asebastudio.exe

../../common/zeroconf/libasebazeroconf.a(zeroconf.cpp.obj): In function `ZNK6Dashel6Stream18getTargetParameterB5cxx11EPKc':
C:/src/dist/windows/include/dashel/dashel.h:330: 
undefined reference to `Dashel::ParameterSet::get[abi:cxx11](char const *) const
```

The call in Aseba::Zeroconf that leads to the linker error is 
https://github.com/aseba-community/aseba/blob/cb260fca437afbc1265763a4d9076ac41e6df0c7/common/zeroconf/zeroconf.cpp#L337
which is implement in Dashel by
```
const std::string& getTargetParameter(const char* param) const { return target.get(param); }
```
I admit that the Aseba::Zeroconf code should use `int Dashel::ParameterSet::get<int>(char const*) const` but :warning: fixing it at this stage would **mask the linker error** and I would prefer to fix that first.

davidjsherman, 2018-01-18:
It's a regression because this breaks the previous Jenkinsfile, which recompiled everything using the default configuration.

For example, this is why many of @cor3ntin's branches [spuriously fail in CI](https://ci.inria.fr/inirobot/job/Corentin/job/aseba/). (They are branches and not merged PRs, and thus use the previous strategy of recompiling Dashel & Enki in the old Jenkinsfile.)

stephanemagnenat, 2018-01-18:
Should we configure Dashel to be built as static library on Windows?

cor3ntin, 2018-01-18:
I'm not sure it's a linker error since the gcc state that the abi should be cxx11 no matter the version of  `-std`  used to compile Dashel. unless mingwin does something to alter the default behaviour.

We can try to compile dashel in c++11 mode and see whether that improves things.

davidjsherman, 2018-01-18:
Here is a comparison of the symbol tables generated with the four combinations on Windows: [dashel-symbol-tables.txt](https://github.com/aseba-community/aseba/files/1642564/dashel-symbol-tables.txt). Curiously I find what I think is the missing symbol:
```
build/SH-OFF_ABI-1/libdashel.a[dashel-win32.cpp.obj]:   Dashel::ParameterSet::get[abi:cxx11](char const*) const U         
build/SH-OFF_ABI-1/libdashel.a[dashel-common.cpp.obj]:  Dashel::ParameterSet::get[abi:cxx11](char const*) const T 00000ce4 
build/SH-ON_ABI-1/libdashel.dll:                        Dashel::ParameterSet::get[abi:cxx11](char const*) const T 67984080 
build/SH-ON_ABI-1/libdashel.dll.a[d000326.o]:           Dashel::ParameterSet::get[abi:cxx11](char const*) const T 00000000 
```

davidjsherman, 2018-01-18:
Looking at https://ci.inria.fr/inirobot/job/Corentin/job/aseba/job/fix_organization_name/6/, for example, the line matching ”/windows/include/dashel/dashel.h:335“: the undefined reference is to
`Dashel::ParameterSet::get[abi:cxx11](char const *) const`, which is indeed undefined in ../../common/zeroconf/libasebazeroconf.a(zeroconf.cpp.obj):
```
$ cd /c/builds/workspace/aseba_fix_organization_name-DUGMGLRR5VY5ZVD33AXOHKF5QHYFA2G36V6OHGTLJZM7XOHYOS6A/build/windows
$ nm -f posix -A -C -a $(find . -name libasebazeroconf\*) | grep -P 'Dashel::ParameterSet::get\W'
./common/zeroconf/libasebazeroconf.a[zeroconf.cpp.obj]: Dashel::ParameterSet::get[abi:cxx11](char const*) const U         
./common/zeroconf/libasebazeroconf.a[target.cpp.obj]: Dashel::ParameterSet::get[abi:cxx11](char const*) const U
```
But it seems to exist in the Dashel .dll library:
```
$ cd ../dashel/windows/
$ nm -f posix -A -C -a $(find . -name libdashel\*) | grep -P 'Dashel::ParameterSet::get\W'
./libdashel.dll: Dashel::ParameterSet::get[abi:cxx11](char const*) const T 67984080 
./libdashel.dll: bool Dashel::ParameterSet::get<bool>(char const*) const T 6798b204 
./libdashel.dll: bool Dashel::ParameterSet::get<bool>(char const*) const U         
./libdashel.dll: double Dashel::ParameterSet::get<double>(char const*) const T 6798b444 
./libdashel.dll: float Dashel::ParameterSet::get<float>(char const*) const T 6798b690 
./libdashel.dll: int Dashel::ParameterSet::get<int>(char const*) const T 6798b8dc 
./libdashel.dll: int Dashel::ParameterSet::get<int>(char const*) const U         
./libdashel.dll: unsigned int Dashel::ParameterSet::get<unsigned int>(char const*) const U         
./libdashel.dll: unsigned int Dashel::ParameterSet::get<unsigned int>(char const*) const T 6798bb1c 
./libdashel.dll.a[d000331.o]: unsigned int Dashel::ParameterSet::get<unsigned int>(char const*) const T 00000000 
./libdashel.dll.a[d000330.o]: int Dashel::ParameterSet::get<int>(char const*) const T 00000000 
./libdashel.dll.a[d000329.o]: float Dashel::ParameterSet::get<float>(char const*) const T 00000000 
./libdashel.dll.a[d000328.o]: double Dashel::ParameterSet::get<double>(char const*) const T 00000000 
./libdashel.dll.a[d000327.o]: bool Dashel::ParameterSet::get<bool>(char const*) const T 00000000 
./libdashel.dll.a[d000326.o]: Dashel::ParameterSet::get[abi:cxx11](char const*) const T 00000000
```
So I don't understand why there is a linker error.

cor3ntin, 2018-01-18:
> fix_organization_name/6/, for example, the line matching

It's mysterious indeed.
Does that happens in a particular branches of aseba and/or dashel ?
We do not have this issue on the EPFL build system and are able to build
the windows installer without issue.

davidjsherman, 2018-01-18:
This happens with the current master branches.

The job [aseba-community/master/shared-on](https://ci.inria.fr/inirobot/job/aseba-community/job/aseba/job/master/46/) (see the [console log](https://ci.inria.fr/inirobot/job/aseba-community/job/aseba/job/master/46/console)) is a parameterized build of Aseba master using my job [David Sherman/dashel/shared-on](https://ci.inria.fr/inirobot/job/David%20Sherman/job/dashel/job/shared-on/), which is Dashel master compiled with BUILD_SHARED_LIBS:BOOL=ON.

About ¾ of the way through the log you will see:
```
[Compile on windows] ../../common/zeroconf/libasebazeroconf.a(zeroconf.cpp.obj): In function 
[Compile on windows] `ZNK6Dashel6Stream18getTargetParameterB5cxx11EPKc':
[Compile on windows] C:/builds/workspace/seba-community_aseba_master-INKVPXY3DG47WFJ4GNADL2DG62EU25Q
[Compile on windows] 4UUWZZOLNUTQURMBB2XXA/dist/windows/include/dashel/dashel.h:335: undefined 
[Compile on windows] reference to `Dashel::ParameterSet::get[abi:cxx11
[Compile on windows] ](char
[Compile on windows]  const*) const
[Compile on windows] '
[Compile on windows] collect2.exe: error: ld returned 1 exit status
[Compile on windows] 
[Compile on windows] make[2]: *** [clients/studio/CMakeFiles/asebastudio.dir/build.make:954: 
[Compile on windows] clients/studio/asebastudio.exe] Error 1
[Compile on windows] 
[Compile on windows] make[1]: *** [CMakeFiles/Makefile2:1066: 
[Compile on windows] clients/studio/CMakeFiles/asebastudio.dir/all] Error 2
[Compile on windows] 
[Compile on windows] make: *** [Makefile:141: all] Error 2
```

cor3ntin, 2018-01-24:
Funnily enough, I encountered the issue while refactoring the CMake build system.
Investigating.

cor3ntin, 2018-01-25:
zeroconf depends on dashel, however that dependency was not properly expressed and the libraries where consequently not ordered properly in some link commands.
Te bug is fixed in my cmake refactor branch

cor3ntin, 2018-03-09:
This is fixed by the new build system

closed 2018-03-09 by cor3ntin