aseba-community/aseba #793: Build Script efforts

cor3ntin, 2018-01-23:
Issue to publicly keep track of an ongoing effort to clean up the cmake build scripts

The goals are numerous 
### Improve the reliability and maintainability of the build scripts
- I'm moving the cmake files to 3.x and will make heavy use of the `target__xxxx` commands to limit the number of globally defined variable. It should make the dependency tree more clear and clean. 
- I'm making the files compiler agnostic
- I'm removing configuration or test specific settings and will put them in CTest files.

### Better Packaging and Qt5 support
- I'll be testing whether we can use CPack especially with `windeployqt` and `macdeployqt`.
- I will try to get rid of the `aseba-windows` and `aseba-osx` repositories so we would just have  `aseba` and `aseba-ci-integration`

### Better Usability and Tool-ability
- I'd like to reorganize both the build tree so that all built binary end up in the same location, and the source tree to make it easier to comprehend. 
   - Eventually move the sources to a source subdir
   - Move the third parties to a top level dir, hopefully with proper licence. That concern things like `avahi-compat-libdns_sd`
- Make it possible ( not mandatory ) to manually clone dashel and enki in the aseba source tree to have aseba build them for us, imitating the LLVM model.

### Cross compilation
I'm investigating whether cross compilation would be suitable to build the projects for windows and mac on docker containers.

And, hopefully, not break anything.

stephanemagnenat, 2018-01-23:
I agree with most goals.

> I'd like to reorganize both the build tree so that all built binary end up in the same location,

I am not convinced of that, because if two files have the same names there will be confusion.

cor3ntin, 2018-01-23:
@stephanemagnenat I mean final binaries. there would be a build/bin folder like you would have in a final packaged version. I think that change is pretty straight forward.

davidjsherman, 2018-01-23:
@cor3ntin @stephanemagnenat: if you look at the Jenkinsfiles, you will see a convention for the directories in which artifacts are compiled and locally installed.

davidjsherman, 2018-01-23:
The Jenkinsfiles are very careful never to install anything globally on the build nodes, because arbitrary combinations of versions are possible. So the Aseba Jenkinsfile imports Dashel and Enki artifacts into <code>dist/<em>arch</em>/<em>product</em></code>, convinces CMake to look for them there, and compiles Aseba in <code>build/<em>arch</em></code>. Thus the directory hierarchy for the archived artifacts for Aseba contains the results for all architectures.

cor3ntin, 2018-01-24:
@davidjsherman I'm replying here to your open issues since it's basically a full rewrite an pose interesting design goals. 

Regarding optional dependencies, I'm thinking that we should make more dependencies required than the other way around.

-  At the code level, having optional dependencies is a maintenance burden and make the code less reliable ( since, in practice, we don't test without zero-conf support ) 
- At the build system level it also make things more complicated. How many people actually want to build Aseba without Qt for example ? Without Dashel ?

Is the burden of getting Dashel for the user (which we should simplify anyway ) worth the maintenance cost of managing it's absence ?

So I'm tempted to make more dependencies required.
At the same time, I'm making sure no individual targets links to more than it has to, so building them individually works and does dot link to unnecessary dependencies

davidjsherman, 2018-01-24:
Well, it's nice to base this kind of decision on some kind of analysis. It should be possible to draw a matrix of build environments versus products, to identify which combinations make sense and ought to be supported.

For some purely anecdotal evidence, I build Aseba for embedded devices and ES8 environments; in most cases I couldn't give a murid's fundament about Qt, it isn't even available for my targets. In ES8 I have websockets but no Dashel-supported TCP/IP sockets; I have SDL but no Qt.

cor3ntin, 2018-01-24:
@stephanemagnenat Can you confirm you agree on tree layout modifications ? 

I want to move the sources to `src`, examples and test would stay at the root.
I'd also like to move some docs to `docs`, leaving index.md, contributing.md,and changelog.md at the root, for now ( do we want the changelog on readthe doc ?)

````
android
authors.txt
changelog.md
CMakeLists.txt
CMakeModules
contributing.md
debian
docs
docsource
Doxyfile
examples
Jenkinsfile
license.txt
localization.md
maintainer
menu
readme.md
rpm
src
tests
```

davidjsherman, 2018-01-24:
Could someone share the justification for this reorganization? I feel like I've missed several important episodes, that explain the expected return on this investment.

cor3ntin, 2018-01-24:
First off, it's a small investment (aka, it's mostly done )
Then there will be some new directories added at the root soon (3rd parties, windows & mac package script ) as well as new sources folder and so it's becoming quite messy.

It's also the occasion to fix the examples and tests which should not depend on relative include paths.

Ad while I'm still unsure about what the finished documentation will look like, i think most documentation should be under docs as this is the root of the readthedocs docs.

davidjsherman, 2018-01-24:
Could you be more explicit about the cost of "it becoming quite messy", so that one could understand what the gain is supposed to be?

I'm not thinking of the cost of making the change, which is minor; I'm thinking of the cost of having to adapt people, current tooling, and future development. You must convince your developers that it is worth it. Or we'll all continuing working in forks with the old organization.

stephanemagnenat, 2018-01-24:
I agree with @davidjsherman that the cost of a change must be considered for all project developers and contributors, not just for the person doing the change. In that sense it is important to discuss these things rather than pushing them hastily.

Regarding optional dependencies, in my experience it was not really a problem to maintain. I have been relatively centric in the sense of not pushing optionality strongly (I kept Dashel as a mandatory dependency for example). That said, if CMake files are properly written, it should not be too hard to even have that.

@cor3ntin I have some questions and comments in the proposed reorganised tree:
- I do not see a top-level 3rd party folder, is it missing or did you change your mind about that?
- The android directory could be deleted I think, as it contains what is necessary to compile thymiovpl standalone on Android, and this is superseded by thymio-vpl2.
- At a later stage, `docsource` could be moved in a source directory inside `docs`.

cor3ntin, 2018-01-24:
@stephanemagnenat 

- I created `third_party` after making the move. For now there is only the avahi compatibility layer but I will add more dependencies in the coming weeks, including Catch2 and https://github.com/nlohmann/json
- Noted For android
- Agree for doc-source. But I have a lot of questions regarding the docs

I think dependencies should be considered on a case by case basis. What's missing is a clear summary of what will or will not be build.

However I strongly suggest we make zero conf non-optional. ( ie that all modules depending on zeroconf always have zeroconf support )

cor3ntin, 2018-01-24:
You can ( but don't have to, of course ) follow progress on https://github.com/cor3ntin/aseba/tree/cmake_3x

stephanemagnenat, 2018-01-24:
Two notes about dependencies:
- nlohmann json is already as a dependency in the http-refactor branch.
- Do we really need Catch2? We had it for a while and then removed it.

And one about Zeroconf:
- On Android neither Avahi nor Bonjour is available, and thus Zeroconf will have to be implemented by talking to the Android implementation in Java through JNI. So we need to keep it optional at least for the core of Aseba.

cor3ntin, 2018-01-24:
I know it's used, and when the branch will be merge, I'll move it in
third_party.

I do think we need a unit test framework, Catch2 happens to be easy to use
and relatively lightweigt. but if you want to use something else, fine by
me :)

For zeroconf, I have no issue making it optional to build the project, but
not for the bits that depends on it ( aka no zeroconf = no switch / http /
etc ) Do you think that's reasonable ?
Of course, the good long term solution is to write a JNI wrapper if that's
needed :)

davidjsherman, 2018-01-24:
If you take the time to look at the code, you will see that Switch and Studio can be compiled without Zeroconf, but do have extra abilities if it is present. In the same way, Playground has extra abilities if SDL is present, Studio can provide charts for debugging events if Qwt is present, Medulla can be used to manipulate the bus if Dbus is present, and so on.

The use of CMake and conditional compilation makes Aseba quite flexible: the same source code can be used to build many different products for many different target systems. The reason that I suggested drawing a product matrix, is that you'll need to test a lot of configurations in order to validate your new build system, and probably you'll find that task much easier if you proceed systematically.

cor3ntin, 2018-01-24:
My point is that it's not realistic.
There are few people working full time on aseba. And I think you will agree that the quantity of tests is limited.

So even if we had CI for all configurations, it would not be a good indication that the product works.
There are also design considerations : *how* should the switch work without zero conf ? How should the launcher work ? Will that be a good UX ?

But more fundamentally, given Zeroconf, QWT, SDL, XML, Various Qt modules, We are looking at something like Maybe 64, 128 possible configurations , on 3 Os. And that's not exhaustive.

If we are pragmatic, all shipped versions will (and do) depend on all these things.

Note that there is a difference between conditionally building a module and conditionally depending on something.
 
For example, Medulla is the same whether dbus is present or not, so we just have to build it once to assert that it builds -  and hopefully, we can test it. And the end result is that you get Medulla or you don't. So that's fine.
And I don't care about worrying of medusa being present or not in the packages since the packages we maintain unconditionally have it. 

However, the switch depending on zero-conf conditionally is much more problematic. We have multiple code-paths, so we need to test them all, reason about them all. Since `#define` are terrible, the code may not even compile and we would never know.

We don't even have the ability to test any of these configurations at the moment. We don't have a practical way to tell cmake to not depend on some dependency ( aka -DNO_ZERO_CONF or something)

davidjsherman, 2018-01-24:
I don't understand. It already works. Why do you insist on adding regressions?

cor3ntin, 2018-01-25:
I'm having CMake build a  package of feature summary so we understand exactly what will or will not be built.
Each feature will have an associated option that can be used to disable the feature at configure time, independently of the required package availability.

```
-- The following packages have been found:

 * Qt5OpenGL
 * OpenGL
 * PythonLibs
 * Boost
 * Doxygen
 * Qt5Core
 * Qt5Gui
 * Qt5Widgets
 * Qt5Network
 * Qt5Sql (required version >= 5.9.1)
 * Qt5Help
 * Qt5Concurrent
 * Qt5Xml
 * Qt5Svg
 * Qt5X11Extras
 * Qt5Test
 * Qt5 (required version >= 5.6)
 * Qt5DBus (required version >= 5.6)
 * DNSSD
 * SDL2
 * Threads
 * Protobuf
 * LibXml2
 * PythonInterp

-- The following packages have not been found:

 * Qwt

-- The following features have been enabled:

 * DISCOVERY, Use AVAHI/DNSSD to discover compatible devices and their status

-- Configuring done
-- Generating done
-- Build files have been written to: /home/cor3ntin/dev/mobsya/aseba-build-linux/aseba

````

cor3ntin, 2018-02-05:
See also : PR #798

cor3ntin, 2018-02-15:
@mbonani  Things are going a bit slow but they are going

- I have the build script in a somewhat clean state, there are still some work to do but i'm reluctant to modify things further until we merge
- I setup a travis project to build a linux version as well a a mac version  https://travis-ci.org/cor3ntin/aseba
That can be setup to be `https://travis-ci.org/aseba-community/aseba` or something.
The build script is committed in the repository and it's pretty straight forward https://github.com/cor3ntin/aseba/blob/travis/.travis.yml

- I've managed to build Aseba to Visual studio and clang and windows. For that I have setup a appveyor instance, same kind of thing https://ci.appveyor.com/project/cor3ntin/aseba
The configuration is committed to https://github.com/cor3ntin/aseba/tree/appveyor

The windows builds are not completely done, they don't use the SDL and qwt yet. They don't use Libxml either but i'd rather switch out to another lib that's more windows friendly.
Some thest are not built because of a get-opt dependency.
And they don't build dynamically

I've found that both tools are significantly faster than Jenkins, especially the windows builds.
The mingw versions still take 20 minutes each because we have to install Qt before each build which takes a while... But we could get rid of the mingw builds all together, at least for the purposes of checking the PR

Work on packaging will be done next week.
Then when everything is merged, I will cleanup the headers installation, translation generation, doc generation.

i was considering using the github release feature to create releases. and push the generated installers.
We should also be able to push directly to launchpad as to generate the `.deb`  for both nighties and releases.

I will also look into some quality assurance tools