aseba-community/aseba #794: Unguarded dependency on QGLWidget

davidjsherman, 2018-01-23:
Compiling Dashel, Enki, and Aseba **without Qt**, targets/playground/asebasim/robots/e-puck fails trying to include QGLWidget through Enki Viewer:
```
[ 71%] Building CXX object targets/playground/CMakeFiles/asebasim.dir/robots/e-puck/EPuck.cpp.obj
In file included from /work/aseba/targets/playground/robots/e-puck/../../PlaygroundViewer.h:33:0,
                 from /work/aseba/targets/playground/robots/e-puck/EPuck.cpp:23:
/work/enki/viewer/Viewer.h:38:21: fatal error: QGLWidget: No such file or directory
compilation terminated.
```
Without Qt, no compilation of a component with a Qt dependency should be attempted.

cor3ntin, 2018-01-24:
Also, QGLWidget is completely deprecated.

stephanemagnenat, 2018-01-24:
> Also, QGLWidget is completely deprecated.

Yes but that's unrelated.

cor3ntin, 2018-01-24:
@stephanemagnenat I didn't try to imply it was, merely a passing remark !