aseba-community/aseba #797: Dependencies Cleaning

cor3ntin, 2018-01-24:
There is a number of dependencies we could get rid of or replace, given the occasion

- libxml2  => QtXml
- SDL2    => QtGamepad ( since Qt 5.7 )
- QWT    => QtCharts ( since Qt 5.7 )

davidjsherman, 2018-01-24:
No, then that introduces a dependency on Qt that is not needed for many products. Asebahttp uses libxml2, Asebacompiler.js uses the SDL provided by Emscripten.

I feel like I am repeating myself, I said essentially the same thing in https://github.com/aseba-community/aseba/issues/793#issuecomment-360063972!

stephanemagnenat, 2018-01-24:
I agree with David, libxml2 is a lower-level library that is well suited for embedded, and SDL2 is also much smaller than Qt and more suitable for command-line tools.

I'm fine with getting ride of Qwt if QtCharts brings similar features or better, but as it currently works well, I do not see the value of changing as there is non-trivial code rewrite to do. Do we have a comparison of the two feature-wise?

cor3ntin, 2018-01-24:
My argument was that if Qt is already installed, QChart is more likely to be installed to. especially on Mac and Windows

I don't have a comparison but I think QChart would be a suitable replacement considering the use we have of it.  
https://doc.qt.io/qt-5.10/qtcharts-overview.html

davidjsherman, 2018-01-24:
Marcel Dassault reputedly said that when designing a new version of an airplane, you change the flight surfaces, the motor, the weapons — but for each model, **change only one at a time**. My strong recommendation is that, if you must redo the build system, then do only that. Make it functionally equivalent to the old one, test it to make sure that there are no regressions, send it out for beta test and sit on it for a week.

If it works and is accepted, then it will be possible to leverage it to add new features.

cor3ntin, 2018-01-24:
I was not suggesting to do that now.