aseba-community/aseba #799: Aseba Studio looses connection with Aseba Playground on Mac

ypiguet-epfl, 2018-01-26:
Aseba Studio quickly looses the connection with Aseba Playground with Thymio.playground. This is observed on macOS 10.12 with 1.6, but maybe it's a more general problem. It happens after a short while, typically 10-60 seconds, 100% of the time. After the first disconnection, Aseba Studio automatically reconnects successfully and immediately disconnects in an infinite loop. The visible consequence is that the VPL window, if open, closes upon disconnection and reopens upon reconnection, and the list of variables etc. on the left of the main Studio window disappears and reappears.

stephanemagnenat, 2018-01-26:
We observed that problem in the past as macOS is putting to sleep the background processes. We solved it in the release package by adding a flag to the Info.plist

Did you observe this problem with the released packages or your own compilation?

ypiguet-epfl, 2018-01-26:
With the released 1.6, without anything nonstandard (in Aseba Studio, About Aseba Studio, Libraries: Aseba version 1.6.0, build version git-c56c7b8d).

stephanemagnenat, 2018-01-26:
Interesting, maybe there is a bug in the packaging. Did the problem happen with the latest 1.5.x release ? @mbonani @cor3ntin was there any change in the packaging script that could have affected that?

ypiguet-epfl, 2018-01-26:
I've just tried with 1.5.5 and unifr.playground (thymio.playground is new in 1.6, I think). Same issue with the first simulated e-puck (without VPL obviously).

stephanemagnenat, 2018-01-26:
Ok, then we have to investigate, I put it in 1.6.1. Thank you for reporting and for the tests!

mbonani, 2018-01-26:
Normally not, I see this behavior only when you minimize playground window. Perhaps we should not allow to minimze the windows.

ypiguet-epfl, 2018-01-26:
I haven't minimized the Playground window.

ypiguet-epfl, 2018-02-03:
Poking Playground by clicking anywhere in its window will prevent disconnections for a short while. So it must be related to Playground having a snooze and Studio having a too short timeout. One of them should be changed; preferably Studio to avoid the same issue with other clients and because disconnections have really nasty consequences for the user.

mbonani, 2018-03-07:
this could be probably solved with new .plist move to new CI

Zach1993, 2018-03-14:
Just started trying to use Studio and am having this problem. I have Aseba studio V 1.6.0 and OSX 10.13.3.