aseba-community/aseba #800: Improve handling of SVG files

cor3ntin, 2018-01-26:
In quite a few places, we use QSvg to render icons.
While this fine, it does introduce a hard dependency on QSvg. It would be possible to use QIcon instead, that would link to QSvg at runtime, removing that dependency

I think it's a better pattern that having modules such as `asebaaboutqt` be conditionally compiled.

stephanemagnenat, 2018-01-26:
There was a reason I used QSvg directly but cannot remember the details, I think something linked to a problem with HiDPI screens or a limitation of QIcon with SVGs. It's fine for me to switch if the underlying problem has been fixed (which I expect to be the case as we switch from Qt 4.x to 5.x).

cor3ntin, 2018-01-26:
@stephanemagnenat oh yeah, HiDPI support in 4.x was a mess - almost non existent really.
It's better now, i think we would still need to specify a default size in the svg files.

stephanemagnenat, 2018-01-26:
> i think we would still need to specify a default size in the svg files.

That's annoying as we want to use the file at different resolutions. If this imply to keep using QSvg, I would keep using it.

cor3ntin, 2018-01-26:
The default size is only useful if you don't specify it in the application.
It does not constrain the way you use the file, merely how viewers render
it. If the application request a given size, say 64x64, that's what it will
get

stephanemagnenat, 2018-01-26:
Ok then it's fine.