aseba-community/aseba #801: Translation - Turkish

onurkahveci, 2018-01-26:
I would like to translate Aseba to my native-language "Turkish". I can create .ts files or if you can create them I can edit them. What should I do next?

I open the issue, because there is a statement:
Please open an new issue requesting the addition of your language. Once the issue is processed, follow the instructions above.

So I am ready to start translating, you can check my old contributions for checking my translation ability.

stephanemagnenat, 2018-01-30:
Preparing the support and fixing #529 .

onurkahveci, 2018-01-30:
@stephanemagnenat I can copy TS files for any of the languages mentioned in the list for example this file for about section: "qtabout_fr.ts" change the name as "qtabout_tr.ts" and translate it in turkish. Fork it, and send pull requests when they are finished. It would be easier for me if it is ok for you? 

List:
For about boxes: common/about/qtabout_LL.ts
For Studio: clients/studio/asebastudio_LL.ts and clients/studio/compiler_LL.ts
For Thymio Firmware Updater: clients/thymioupgrader/thymioupgrader_LL.ts
For Thymio Wireless Configurator: clients/thymiownetconfig/thymiownetconfig_LL.ts
For Aseba Playground: targets/challenge/asebaplayground_LL.ts
For Aseba Challenge: targets/challenge/asebachallenge_LL.ts Where LL is the ISO 639-1 code of your language.

stephanemagnenat, 2018-01-30:
@onurkahveci I am updating the automatic generation script, new translations are a good excuse to update it, give me a bit of time and you'll get the files...

stephanemagnenat, 2018-02-02:
I have added Türkish support with empty translations in 02db54f5b29c78c489e00f6cfb6fb2fa8deaa620. You can take the corresponding files directly from the tree and start translating. Thank you very much for your help!

onurkahveci, 2018-02-04:
Thank you. I will start translation tonight. Have a nice day!

onurkahveci, 2018-02-13:
@stephanemagnenat turkish translation file of studio is added, can you please check and merge it. I will add other files soon. Thanks.
PR: https://github.com/aseba-community/aseba/pull/836

stephanemagnenat, 2018-03-01:
PR merged.

closed 2018-04-17 by mbonani