aseba-community/aseba #804: Aesl file created in VPL standalone cannot be used in Studio

ypiguet-epfl, 2018-01-30:
If you create a program in Thymio VPL, you save it as a .aesl file and you try to open it in Aseba Studio connected to the same Thymio node, Studio displays _1 scripts have no corresponding nodes in the current network and have not been loaded_ (sic). The soure code is displayed in a new tab named _thymio-II (not available)_ without the node-specific controls on the left of the window. Tested in 1.6, macOS 10.12, Thymio simulated in Playground. See also #670.
Suggestion 1: if you can't match the node id, have a button to match the node by name if it's possible ("thymio-II" here).
Suggestion 2: check the grammar in error messages.

stephanemagnenat, 2018-01-30:
@cor3ntin could you please try to reproduce this?

cor3ntin, 2018-02-02:
No reproduction on linux, I'll investigate later.

mbonani, 2018-03-10:
The problem is that when you open a aesl that contain blockly information it crash studio because it want to open blockly that was remove. It works on Linux because Blockly is still there. 
Generate files have also strange info not needed. <xml xmlns="http://www.w3.org/1999/xhtml">
[untitled (8).zip](https://github.com/aseba-community/aseba/files/1799867/untitled.8.zip)

cor3ntin, 2018-03-12:
I fixed the crash the file opens now fine in Aseba.
However when the file is saved, the parts specific to the missing plugin are lost.
I guess fixing the crash is a reasonable approach for 1.6

closed 2018-03-15 by cor3ntin