aseba-community/aseba #805: Sqrt native function does not check for negative input values

stephanemagnenat, 2018-01-30:
This issue was reported by Alex Brodsky.

ypiguet-epfl, 2018-02-01:
Confirmed. The first loop doesn't terminate for a negative argument in [vm/natives.c](https://github.com/aseba-community/aseba/blob/master/vm/natives.c) at line 233-234. The fix is trivial.

Exactly what's computed should be documented. For `sqrt`, it must be `floor(sqrt(x))`, arguably less useful than `round(sqrt(x))`. For `sin`, it's close to `floor(32767*sin(x*pi/32768))`, but not quite (e.g. `sin(7)` gives `22` while `sin(7*pi/32768)*32767` is approximately 21.9905.

stephanemagnenat, 2018-02-05:
The cleanest would be a message with name `DomainError` (the equivalent of `std::domain_error`).

stephanemagnenat, 2018-02-05:
Another "hack" would be to rename `DivisionByZero` to `DomainError`, and interpret the type of error by looking up which pc generated the error.

stephanemagnenat, 2018-02-05:
For now I used the `NodeSpecificError` message. This does not allow to tell which variable of the array has the negative value, nor is translated. I created a new issue for that #820.