aseba-community/aseba #813: Disjoint angle ranges in VPL accelerometer event

ypiguet-epfl, 2018-02-03:
In the code generated for the accelerometer event in advanced mode, the acceleration angle is divided into ranges of 15 degrees. But these ranges are disjoint, because strict inequality comparisons are used with the limit angle. For instance we can get
```
onevent acc
  call math.atan2(angle, acc[0], acc[2])
  when angle > -1365 and angle < 1365 do
    motor.left.target = 0
    motor.right.target = 0
    emit pair_run 0
  end
  call math.atan2(angle, acc[0], acc[2])
  when angle > 1365 and angle < 4095 do
    motor.left.target = 0
    motor.right.target = 0
    emit pair_run 1
  end
```
This leaves a hole (`angle==1365`) which can give an unexpected and problematic behavior if a larger range is built from two pieces as in the code above.

Suggestion: use >= for the lower limit and < for the higher limit in method `Compiler::CodeGenerator::visitEventAcc` of [CompilerVisitors.cpp](https://github.com/aseba-community/aseba/blob/master/clients/studio/plugins/ThymioVPL/CompilerVisitors.cpp)

davidjsherman, 2018-02-03:
I was trying to write a failing test for the current code, a VPL program that fails to emit any `pair_run` event at all, if acc[0]=200 and acc[2]=1521 for example, in PITCH mode. But the results of `atan2` are discretized in steps of 5, and the closest I ever came was an angle of 1364. Do you have an example where the situation you describe actually arises?

I do agree with your suggestion to change https://github.com/aseba-community/aseba/blob/cdae20d330ee705e189a81d060d602603e9ff29d/clients/studio/plugins/ThymioVPL/CompilerVisitors.cpp#L629
but it would have been nice to be able to verify that the change fixes something.

stephanemagnenat, 2018-02-03:
Thanks @ypiguet-epfl for finding the bug with the angles. I put it as a bug for milestone 1.6.1.

I agree that the implementation of `atan2` is simplistic. If there is a better version, for example that uses less memory space for the same runtime, or has better runtime for the same memory space, I'll be glad to accept a PR. Note that the new version should come with full test coverage because we do not want to risk to break `atan2`.

ypiguet-epfl, 2018-02-04:
For a PR, after #782 I won't volunteer for a while.

Ideally test coverage should be done anyway on parts deemed to worth it, not as an additional hindrance to potential improvements.

stephanemagnenat, 2018-02-04:
> Ideally test coverage should be done anyway on parts deemed to worth it, not as an additional hindrance to potential improvements.

I agree, I created a wish issue #814.

commit 1003327d (2018-03-07): Fix #813
commit 8b06d0e9 (2018-03-07): Fix #813
closed 2018-03-07 by cor3ntin