aseba-community/aseba #821: Playground SIGSEV on Windows

davidjsherman, 2018-02-05:
On Windows, commit 861798403b886aae153f9ee2210bd2e16a0457e4, for any playground: some warnings then SIGSEV
```
warning: Vertex shader for simpleShaderProg (MainVertexShader & PositionOnlyVertexShader) failed to compi        le
warning: Fragment shader for simpleShaderProg (MainFragmentShader & ShockingPinkSrcFragmentShader) failed         to compile
warning: QGLShaderProgram: shader programs are not supported
warning: Errors linking simple shader:
warning: Vertex shader for blitShaderProg (MainWithTexCoordsVertexShader & UntransformedPositionVertexSha        der) failed to compile
warning: Fragment shader for blitShaderProg (MainFragmentShader & ImageSrcFragmentShader) failed to compi        le
warning: QGLShaderProgram: shader programs are not supported
warning: Errors linking blit shader:

Thread 1 received signal SIGSEGV, Segmentation fault.
0x00000000 in ?? ()
```

stephanemagnenat, 2018-02-05:
We should try to reproduce the problem with previous commits through bisection, as this one only changes the author file.

cor3ntin, 2018-02-05:
That is probably the first of a long list of Qt-5 related bugs !

davidjsherman, 2018-02-08:
I don't want to belabor this point but currently *master* is broken, now requiring a bisect to discover where in the long list of recently merged commits something went wrong.

davidjsherman, 2018-02-08:
@cor3ntin what can you suggest for headless testing on Windows?