aseba-community/aseba #823: When opening a file in a fresh VPL window, alert to save changes

ypiguet-epfl, 2018-02-06:
In Aseba Studio 1.6, in the VPL window opened immediately after launch, when clicking the Open icon to open an aesl file, an alert is displayed: _The document "Untitled" has been modified. Do you want to save your changes or discard them?_

Suggestion: don't display this alert if there is no change to save.

stephanemagnenat, 2018-02-07:
If the window opens when the AESL file has no VPL code it is a bug, it should not. Does it happen with all AESL files, or only some?

ypiguet-epfl, 2018-02-07:
It's before any aesl file has been opened. To reproduce (tested with 1.6 on macOS 10.12):
- start Aseba Playground with thymio.playground
- start Aseba Studio, connect to the simulated Thymio in Playground
- click the _Launch VPL_ button (bottom left of the Studio main window)
- click the _Open_ button (folder icon, top left of the VPL window)

You get the alert, which for me is a bug.