aseba-community/aseba #824: Filipino Translation

josephace135, 2018-02-07:
Hi,

I would like to contribute on your project by translating (upon reading the localization guidelines).
I will translate the necessary files to my native language, Filipino.

Best Regards,

josephace135

stephanemagnenat, 2018-02-07:
Hello, thank you for contributing to Aseba! Could you please give me the exact name of the language (in English and natively) and the [ISO 639-1](https://en.wikipedia.org/wiki/List_of_ISO_639-1_codes) code to use? As there seems to be [several languages](https://en.wikipedia.org/wiki/Filipinos#Languages).

josephace135, 2018-02-08:
Hello @stephanemagnenat, as for ISO 639-1, only Tagalog is listed with code of tl.
Does this support ISO 639-2 too?

stephanemagnenat, 2018-02-08:
The Aseba translation support is built on [Qt](http://doc.qt.io/qt-5/internationalization.html). According to the documentation of [`QLocale::name()`](http://doc.qt.io/qt-5/qlocale.html#name):

> Returns the language and country of this locale as a string of the form "language_country", where language is a lowercase, two-letter ISO 639 language code, and country is an uppercase, two- or three-letter ISO 3166 country code.

Does your language fit into this scheme?

josephace135, 2018-02-08:
Hello @stephanemagnenat, 

In general, tagalog or tl or tgl for Tagalogs. And then Filipino is the official language of philippines.

stephanemagnenat, 2018-02-08:
So should we use `tl_PH`?

josephace135, 2018-02-08:
Well, looks like tl_TL, fil_PH?

stephanemagnenat, 2018-02-09:
Ok, I'll go with `tl_TL` for the moment.

stephanemagnenat, 2018-02-09:
After a survey of locale support online, I went for simply `tl` with the Filipino (see f3775578744fc5683e6743b87703c5c817fff4f9).