aseba-community/aseba #826: Enforcing the contribution guidlines

cor3ntin, 2018-02-07:
We should prevent contributors ( including core contributors ) to bypass review and CI on master.
Non-forward push should also never ever be done, security concerns non-withstanding.

This can be setup in `Settings > Branches > Master`

[image]

stephanemagnenat, 2018-02-07:
I agree, we should be less lazy ;-)

Non-forward push can be used in extreme case for example to correct a mistake, but we can prevent it and temporarily change the settings should the need arise.

davidjsherman, 2018-02-07:
I really don't like to use technical solutions for social issues; I have confidence in human beings. It should be enough that we agree to wait until review before merging. That is the spirit of the guidelines that I proposed.

cor3ntin, 2018-02-07:
@stephanemagnenat I disagree. I'd rather have a fixing commit on master that a screwed up history. Mistake happen, no need to hide them. Of course that's only true for public/shared branches & tags. Once again, if someone commits a password or something, there can be exceptions ( hopefully, with reviews, that never happens)

@davidjsherman hope for the best, prepare for the worst :)
Regardless of people intentions, some contributors don't know git enough to not screw up unwillingly, we should help them not make mistakes !

closed 2018-04-17 by cor3ntin