aseba-community/aseba #829: Aseba/VPL not starting when trying to use with simulator on Mac 10.3.3.

tiemes, 2018-02-08:
On Mac 10.13.3 with Aseba 1.6.0:
If trying to start Aseba Studio with a simulated Thymio it freezes without opening GUI. I can run Playground and select a file. I see the selected environment but get error messages in the print below. (see screenshot). On top of the Mac screen the "VPL" oder "Aesba Studio" (tried both) name appears but the cursor keeps rotating and no window opens.

[image]
[image]

cor3ntin, 2018-02-08:
could you maximize the playground window and take another screenshot so we can see all the yellow text ?
Thanks

mbonani, 2018-02-08:
It's perhaps related to zeroconf bonjour.

davidjsherman, 2018-02-08:
The red message claims that a process has crashed. In the “Otter” playground, the only additional process that is run is the one that loads the Thymio default behaviors.

tiemes, 2018-02-09:
I'll try to get the a larger window screenshot. It's not my maschine so it may take a while.

cor3ntin, 2018-02-11:
@tiemes never mind, I do have a reproduction for this issue and will try to have a fix shortly. thanks for reporting it !

tiemes, 2018-02-13:
Apparently the connection works albeit the error message in the simulator. Our problem was, that we did not change from serial to tcp mode in the connector window. This procedure is not explained anywhere and one would not expect such a menu behind the info-icon button.
[image]

mbonani, 2018-02-19:
There is no need to change on this menu. Normally selecting by clicking on the discovered device rewrite the correct  connection command. Double click is normally also working.