aseba-community/aseba #840: Evènement manquant

Thyrf, 2018-02-21:
Bonjour,

je programmais un activité avec plusieurs Thymio et j'ai été surpris de voir que certains evènements ne semblait pas être envoyés ou reçu. J'ai donc fait un petit test avec deux Thymio comme voici:

Code pour le premier Thymio:

var counter = 0
timer.period[0]= 500

onevent timer0
	emit increment1 counter
	emit increment2 counter
	counter++

Code pour le deuxièmeThymio:

var counter1 = 0
var counter2 = 0
onevent increment1
	counter1++
onevent increment2
	counter2++

Quand je lance le programme et que je regarde les evènements sur l'ensemble du réseau, je vois cela : 

[image]

Ou on voit que certains evènements n'ont pas été envoyé ou reçu, ici les evènements increment1 avec pour paramètre 45 puis 46.

J'utilise Aseba 1.6.0 et mes robots ont le dernier firmware.

PS: j'ai aussi parfois un problème avec les boutons Charger tout et Exécuter tout ou certains robots ne recoivent pas la commande, mais cela me pose moins problème que le problème d'évènements.

mbonani, 2018-02-21:
Je suppose que c'est avec des Thymio Wireless. 
En effet les message utilisateur ne sont pas garantie par les modules Wireless. On ne voulait pas bloquer un réseau si par exemple un robot s'éloigne trop et que la radio ne passe plus. C'est pas idéal mais pour l'heure difficile de changer cela. 
Si suivant ton application tu veux avoir quelque chose de plus fiable, il faut implémenter un handshake.

Thyrf, 2018-02-21:
Mmh, ok je comprends.

Merci Michael.

FrancescoMondada, 2018-02-21:
Est-ce qu'on ne considère pas cela comme un bug? Est-ce que c'est documenté? Est-ce que cela apparaît seulement (ou beaucoup plus fréquemment) quand il y a deux roots sur le même réseau (ce que j'imagine) On sait aussi que cela se passe uniquement avec le dongle et pas entre les robots, si je ne me trompe pas (peux-tu vérifier @Thyrf ?), ce qui rend cela encore plus bizarre.

Thyrf, 2018-02-21:
Je ne suis pas sur de bien comprendre la difference que tu fais entre communication entre robots et celle avec le dongle. 

Si je ne me trompe tu penses que seul le dongle/aseba ne reçoit pas l'évènement increment1 ? Si c'est le cas, je ne pense pas que cela ne se passe que dans la comm avec le dongle, car je vois le décalage grandir entre counter1 et counter2, ce qui voudrait dire que le robot ne reçoit pas non plus l'évènement.

Thyrf, 2018-02-28:
Après avoir parlé avec plusieurs personnes de ce problème, la majorité semble s'accorder sur l'idée qu'il serait normal d'avoir un problème de communication si les robots sont éloignés, et donc également à ce que le réseaux soit bloqué dans si un robot s'éloigne trop des autres. Ils semblent aussi s'accorder sur le fait qu'il est beaucoup moins compréhensible et acceptable d'avoir des robots qui perdent des événements alors qu'ils sont très proches...