aseba-community/aseba #845: Imported documentation has authorship problems

stephanemagnenat, 2018-02-28:
The imported documentation from the Wiki should retain the list of authors for each page. Simply summarizing the names extracted from the wiki per page should be enough. Also the license should be mentioned as being _Creative Commons Attribution-ShareAlike 3.0_ as stated by the Wiki. The documentation copyright should mention the authors, not me or Mobsya (we would be part of the authors obviously).

stephanemagnenat, 2018-03-01:
After some thinking I believe the best is to change the copyright notice to mention the Aseba authors and contributors, and put the name of people who translated the documentation in the main `authors.txt` file under the contributor section, as there are little original contributions to the documentation from people outside the core Aseba contributors.

cor3ntin, 2018-03-01:
@stephanemagnenat can you handle it ?