aseba-community/aseba #846: Move doxygen comments of public declarations from cpp to header files

ypiguet-epfl, 2018-03-06:
At some places such as in [aseba/compiler/compiler.cpp](https://github.com/aseba-community/aseba/blob/master/compiler/compiler.cpp) for method `compile`, the doxygen documentation of public methods is located in the cpp file. It would be better to put it in the header file ([aseba/compiler/compiler.h](https://github.com/aseba-community/aseba/blob/master/compiler/compiler.h)), because the cpp file could be provided as compiled code and not readily available. That's already the case for some of the public declarations (e.g. `struct SubroutineDescriptor`).

cor3ntin, 2018-03-06:
Would you be willing to make a PR ?

ypiguet-epfl, 2018-03-06:
I think the git/github work wouldn't be worth the trouble for someone like me whose PR are limited to minor code polish. This can wait.

stephanemagnenat, 2018-03-06:
For information, this was a design choice: at the time the compiler was written, code folding wasn't very widely available in editors and so the idea was that either someone knows more or less the code and wants to get a quick look at the available functions, and thus reads the h file, or this person wants the detailed information, and reads the generated documentation. This rationale might still be valid for people using editors such as vim.