aseba-community/aseba #869: a build error

zodiac911, 2018-03-19:
`[ 63%] Building CXX object aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o
In file included from /home/nwpu/aseba/aseba/aseba/clients/studio/MainWindow.h:55:0,
                 from /home/nwpu/aseba/aseba/aseba/clients/studio/StudioAeslEditor.cpp:21:
/home/nwpu/aseba/aseba/aseba/clients/studio/HelpViewer.h:23:30: fatal error: QtHelp/QHelpEngine: No such file or directory
compilation terminated.
aseba/clients/studio/CMakeFiles/asebastudio.dir/build.make:298: recipe for target 'aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o' failed
make[2]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o] Error 1
CMakeFiles/Makefile2:2080: recipe for target 'aseba/clients/studio/CMakeFiles/asebastudio.dir/all' failed
make[1]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2`
any thoughts?

cor3ntin, 2018-03-20:
We have not updated the build documentation yet. This should help you resolve your issue ! Please report back :) https://github.com/cor3ntin/aseba/blob/building_md/docs/en/development/building.rst

zodiac911, 2018-03-20:
@cor3ntin I followed the instruction here https://github.com/aseba-community/aseba/blob/master/docs/compile.Linux.md
and I didn't see any difference coming in the new one, but I'll give it a try though.

cor3ntin, 2018-03-20:
@zodiac911 I'm sorry that you are hitting this issue. I'll fix them tomorrow. Can you tell me more aout your system or attach `/home/nwpu/aseba/aseba/build/CMakeFiles/CMakeOutput.log` and `/home/nwpu/aseba/aseba/build/CMakeFiles/CMakeError.log`  ? Thanks

zodiac911, 2018-03-21:
sorry there is no error file there, I'll paste the content of outputlog here.
```
The system is: Linux - 4.4.0-21-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/nwpu/aseba/aseba/build/CMakeFiles/3.5.1/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/nwpu/aseba/aseba/build/CMakeFiles/3.5.1/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_c835b/fast"
/usr/bin/make -f CMakeFiles/cmTC_c835b.dir/build.make CMakeFiles/cmTC_c835b.dir/build
make[1]: Entering directory '/home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_c835b.dir/testCCompiler.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_c835b.dir/testCCompiler.c.o   -c /home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_c835b
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c835b.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_c835b.dir/testCCompiler.c.o  -o cmTC_c835b -rdynamic 
make[1]: Leaving directory '/home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_3e723/fast"
/usr/bin/make -f CMakeFiles/cmTC_3e723.dir/build.make CMakeFiles/cmTC_3e723.dir/build
make[1]: Entering directory '/home/nwpu/aseba/aseba/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_3e723.dir/CMakeCCompilerABI.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_3e723.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.5/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_3e723
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_3e723.dir/link.txt --verbose=1
/usr/bin/cc      -v CMakeFiles/cmTC_3e723.dir/CMakeCCompilerABI.c.o  -o cmTC_3e723 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.3.1-14ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.3.1 20160413 (Ubuntu 5.3.1-14ubuntu2) 
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/5/:/usr/lib/gcc/x86_64-linux-gnu/5/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/5/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/5/:/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/5/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTC_3e723' '-rdynamic' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/5/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/5/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper -plugin-opt=-fresolution=/tmp/ccoSZXiD.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --sysroot=/ --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -z relro -o cmTC_3e723 /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/5/crtbegin.o -L/usr/lib/gcc/x86_64-linux-gnu/5 -L/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/5/../../.. CMakeFiles/cmTC_3e723.dir/CMakeCCompilerABI.c.o -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /usr/lib/g [... cut ...]
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates
```

cor3ntin, 2018-03-21:
@zodiac911 what distribution do you use ? version of cmake ? What serie of commands did you type ?

zodiac911, 2018-03-21:
@cor3ntin sorry for my carelessness, I'ts just as you said. I tried to build aseba/aseba, and I tried to build the correct directory just now, the exact error I mentioned from the beginning happened again.
```
[ 63%] Building CXX object aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o
In file included from /home/nwpu/aseba/aseba/aseba/clients/studio/MainWindow.h:55:0,
                 from /home/nwpu/aseba/aseba/aseba/clients/studio/StudioAeslEditor.cpp:21:
/home/nwpu/aseba/aseba/aseba/clients/studio/HelpViewer.h:23:30: fatal error: QtHelp/QHelpEngine: No such file or directory
compilation terminated.
aseba/clients/studio/CMakeFiles/asebastudio.dir/build.make:298: recipe for target 'aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o' failed
make[2]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/StudioAeslEditor.cpp.o] Error 1
CMakeFiles/Makefile2:2080: recipe for target 'aseba/clients/studio/CMakeFiles/asebastudio.dir/all' failed
make[1]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

should I remove some stuff?
By the way I use Ubuntu 16.04 cmake version 3.5.1. and the commands I used is what I copied from your instruction

cor3ntin, 2018-03-21:
you are probably missing the packagers `qttools5-dev-tools` or `qttools5-dev`

zodiac911, 2018-03-21:
@cor3ntin I'm sorry, you are right. I tried to install too many dependencies at one time, and the qttools5-dev was not correctly installed. I didn't notice that. But when I run Ctest, some of the tests failed with [Badcommand].
is that a problem?

cor3ntin, 2018-03-21:
@zodiac911 Not really, but I would appreciate if you could share the logs ! thanks

kofa73, 2018-11-24:
> We have not updated the build documentation yet. This should help you resolve your issue ! Please report back :) https://github.com/cor3ntin/aseba/blob/building_md/docs/en/development/building.rst

The link above no longer works. The current link is https://github.com/cor3ntin/aseba/blob/master/docs/en/development/building.rst
In addition to the dependencies listed in the instructions, I also had to install `qttools5-dev`. Finally, I think including `sudo make install` in the instructions would be helpful for those who rarely or never build software themselves. I hope https://launchpad.net/~stephane.magnenat/+archive/ubuntu/bionic will be updated, as the version I obtained from there currently lacks `asebastudio`.