aseba-community/aseba #872: BCU lausanne: instabilité de l'interface Aseba Studio

KevinPradervand, 2018-04-18:
Bonjour, 

Lors de la première connexion du thymio sur Aseba Studio, ce dernier plante au bout de quelques minutes. Nous avons beau redémarrer l'application, l'interface d'Aseba ne s'affiche plus alors que son processus est bien visible sur le gestionnaire de tâche Windows. Aseba n'affiche que la sélection du port série permettant de connecter le Thymio et ne va pas plus loin.

Nous travaillons sur un système VDI composé de Zero-client Samsung NC241 connectés à des machines virtuelles (VMware) Windows 7(64bits)  gérée par Horizon View.

les appareils Thymio ou dongle wifi sont bien détectés et les drivers s'installe correctement, toutefois l'application "freeze" au bout d'un certain temps d'utilisation et ne s'affiche plus par la suite par contre le service continue à fonctionner.

Merci de nous éclairer sur ce sujet, nous avons fait tout les test possible de notre côté et tout semble correcte.

La BCULausanne a déjà acheté un certain nombre de Thymio dans le but de former nos bibliothécaire.
Au final nous voudrions ajouter à nos services usager, la possibilité d'emprunter des Thymio. 

Merci et bonne journée

Kevin Pradervand 
Informaticien BCU Lausanne

mbonani, 2018-04-18:
Bonjour,
Si l'application a freezer il se peut qu'elle n'a pas libérer le port correctement. Pour relancer il faut alors débrancher le robot et contrôler que le logiciel ne tourne plus dans les tâches.
Sinon il y a la routine  "Redémarrer le port USB" dans Aseba->Outils qui aide parfois dans ce genre de cas.
Par contre pourquoi l'application freeze après un certain temps il nous faudrait plus d'information? Y a t il un  message d'erreur ou quelque chose à faire de reproductible? On a aussi peut d'expérience avec votre genre d'installation. 
Meilleures salutations

mbonani, 2018-05-01:
Bonjour,
Avez vous vu des améliorations?
meilleures salutations