aseba-community/aseba #874: Asebamedulla missing in build

richban, 2018-04-18:
I've successfully managed to build `Aseba` under `Mac OS X; 10.13.3; x86_64` according to the [docs ](https://github.com/aseba-community/aseba/blob/master/docs/en/development/building.rst). However I can't find the `asebamedulla` in the build directory.

```
aseba-qt-gui			asebaplayground
aseba-test-invalid-utf8		asebarec
aseba-test-msg			asebashell
aseba-test-natives-count	asebastudio
aseba-test-simulator		asebaswitch
asebabotspeak			asebatargetadvertise
asebachallenge			asebatargetlist-dashelhub
asebacmd			asebatargetlist-qt
asebadummynode			asebatargetlist-thread
asebadump			asebatest
asebaexec			rendervplblocks
asebahttp			thymioupgrader
asebahttp2			thymiovpl
asebajoy			thymiownetconfig
asebamassloader			thymiownetconfig-cli
asebaplay			tst_compiler_utf8
```

It is not supported on `macOS` ?

cor3ntin, 2018-04-19:
Indeed, dbus is only supported on linux.
But it should be possible to install dbus through Homebrew, then compile Qt yourself and use that to build aseba, if you absolutely need it.

richban, 2018-04-19:
I had `dbus` already installed as well `qt qt-legacy-formula qt-webkit@2.3 qt@4 qt@5.5 qwt qwt-qt4` with `brew`.

closed 2018-10-20 by richban