aseba-community/aseba #881: Aseba Studio missing in Ubuntu 18.04

ajs54, 2018-09-10:
Hello.
I don't know if this is the right place to signal this issue, but some binaries are missing in the Ubuntu 18.04 package. As the team handling this package did not answered on launchpad (https://bugs.launchpad.net/ubuntu/+source/aseba/+bug/1778362), I submit this problem here.
If no solution is proposed (I may need these binary soon), I will try and compile them (but I'd prefer an other solution - I know what compilation may turn to be).
Thanks in advance for your help.

ajs54, 2018-09-11:
Nice try, but compilation does not work : dashel package not provided...  No result after a web search, I will check if it is already declared or if I have to open a new bug...

ajs54, 2018-09-11:
OK, I understand !  The dashel directory of the zip provided by your site is empty, and it generates a compilation error...  
But I realize I may not have taken the right archive... I though afterwards that I need the 1.6 release, not the master archive, but 1.6 release depends on QT4, which is not available on Ubuntu 18.04.

I as feared, compilation is rarely easy!..

ajs54, 2018-09-11:
Well, I have to check: the compile.Linux.md seems to be the one of the 1.5 release, the compilation may work with QT5 (I will try tomorrow).

stephanemagnenat, 2018-09-12:
Yes the master branch is on Qt5. It should compile, did you do the clone recursively or did you try `git submodule init && git submodule update`?

ajs54, 2018-09-12:
Thank you for your help, Stéphane.
No, I just get the archive of the master: dashel & enki directories are empty in this archive, I had to download their archive (from the corresponding branches).  That step is OK now, but new problems occurred: PythonLibs, Qt5Help & Qt5LinguistTools seems to be missing. I did import all the dependencies given (I have libqt5help, for example)...
Here is the output of cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF ..
-- Git executable found
-- libudev found
Using dashel from /usr/local/share/aseba-master/dashel
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL2: /usr/lib/x86_64-linux-gnu/libSDL2main.a;/usr/lib/x86_64-linux-gnu/libSDL2.so;-lpthread  
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Python version from interpreter is 2.7.15
-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) 
CMake Warning at enki/python/CMakeLists.txt:33 (message):
  Python libs or executable, or Qt5, or OpenGL not found, skipping Python
  bindings

-- SDL2 found, enabling joystick support
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Using Enki from /usr/local/share/aseba-master/enki
-- Found PythonInterp: /usr/bin/python (found suitable version "2.7.15", minimum required is "2.0") 
-- Could NOT find Qt5Help (missing: Qt5Help_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:39 (message):
  Failed to find Qt5 component "Help" config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt5Help/Qt5HelpConfig.cmake"
Call Stack (most recent call first):
  CMakeLists.txt:37 (find_package)

-- Could NOT find Qt5LinguistTools (missing: Qt5LinguistTools_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:39 (message):
  Failed to find Qt5 component "LinguistTools" config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt5LinguistTools/Qt5LinguistToolsConfig.cmake"
Call Stack (most recent call first):
  CMakeLists.txt:37 (find_package)

-- Found Qwt: /usr/lib/libqwt-qt5.so (found version "6.1.3") 
-- Found DNSSD: /usr/lib/x86_64-linux-gnu/libdns_sd.so  
-- Found ZeroConf in /usr/include;/usr/lib/x86_64-linux-gnu/libdns_sd.so
CMake Error at aseba/common/about/CMakeLists.txt:17 (qt5_add_translation):
  Unknown CMake command "qt5_add_translation".

-- Configuring incomplete, errors occurred!
See also "/usr/local/share/aseba-master/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/share/aseba-master/build/CMakeFiles/CMakeError.log".

I will continue to investigate, and give a final summary of what is missing in the building help!

stephanemagnenat, 2018-09-12:
About the archive: I guess this to be a limitation of github not getting submodules in the archive.

About the missing packages, indeed the documentation on Linux compilation is incomplete. I suggest to try installing the following packages: `python-dev`, `qttools5-dev-tools` and `libqt5help5`.

ajs54, 2018-09-14:
OK, I had to add python-dev, libboost-python-dev, doxygen & qttools5-dev (I already had qttools5-dev-tools & libqt5help5) for cmake to work.
Compilation and installation went right, I will test tomorrow.

ajs54, 2018-09-15:
Fine, it works (after a few dialout-like worries, and some disappearance of the VPL window).
I sum up the building steps in a script, I still have to check it and I will post it here.
If anybody can tell me how to generate a correct Debian package, I can try it.

ajs54, 2018-09-15:
OK, it works. All (the 3 archives = master + dashel + enki, the script and a read-me) can be downloaded on my web pages: http://homepages.loria.fr/AScheuer/Robots/Thymio/build_Aseba_Debian.tbz (19.4 Mo, TarBzip file).
I may not have time to find out how to generate the Debian package until a few months. A quick tutorial (or a link to) could be helpful.

closed 2018-09-15 by ajs54