aseba-community/aseba #884: asebastudio crashes when text is typed

Lopezo, 2018-10-10:
I setup asebastudio on my computer. I am using archlinux (https://aur.archlinux.org/packages/aseba-git/). 
I can launch asebachallenge and asebastudio. I can also connect my own Thymio II (either wifi or using the link). The variable are refreshed. It seems to work well.

However, when I try to type something (or even click on the button `onevent` on top the the text field) asebastudio crash with this error message:
```
terminate called after throwing an instance of 'QUnhandledException'
what():  std::exception
Abandon (core dumped)
```
It seems related to QT5. Do I have to downgrade my QT version (`Qt version 5.11.2`)?

stephanemagnenat, 2018-10-10:
I just tested the latest master version with Qt 5.9.3 and could not reproduce the problem you experienced.

Lopezo, 2018-11-05:
Adding the en_US-UTF-8 locale in the file `/etc/locale.gen` and generating locale with `locale-gen` as suggested here https://github.com/aseba-community/aseba/issues/885#issuecomment-433295110 solve the issue

closed 2018-11-05 by Lopezo