aseba-community/aseba #885: Problem to compile a deb package of aseba

crustyourmind, 2018-10-14:
I'm trying to make a debian sid amd64  package for aseba (after reading compil.linux.md)
Works fine to make the deb package enki and dashel
but I've got a problem with the compilation since the source code of aseba
```
ragnarok@valhalla:~/compilation/aseba/aseba$ debuild -i -us -uc -b
....
[100%] Built target aseba-test-simulator
make[2] : on quitte le répertoire « /home/ragnarok/compilation/aseba/aseba/debian/build »
make[1] : on quitte le répertoire « /home/ragnarok/compilation/aseba/aseba/debian/build »
 fakeroot debian/rules binary
install -d debian/tmp/DEBIAN
install -d debian/tmp/usr/bin
install -d debian/tmp/usr/share/doc/aseba
install -d debian/tmp/usr/share/doc/aseba/example-code
install -d debian/tmp/usr/share/doc/aseba/example-scenarios
cp changelog.md debian/tmp/usr/share/doc/aseba/changelog
cp debian/changelog debian/tmp/usr/share/doc/aseba/changelog.debian
cp debian/copyright debian/tmp/usr/share/doc/aseba
cp targets/playground/examples/*.png debian/tmp/usr/share/doc/aseba/example-scenarios
cp: impossible d'évaluer 'targets/playground/examples/*.png': Aucun fichier ou dossier de ce type
make: *** [debian/rules:27: binary-arch] Error 1
dpkg-buildpackage: erreur: fakeroot debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -rfakeroot -us -uc -ui -i -b failed
```
Effectively, if I look the dir I see it's empty 
```
ragnarok@valhalla:~/compilation/aseba/aseba$ ls -R debian/tmp/
debian/tmp/:
DEBIAN  usr

debian/tmp/DEBIAN:

debian/tmp/usr:
bin  share

debian/tmp/usr/bin:

debian/tmp/usr/share:
doc

debian/tmp/usr/share/doc:
aseba

debian/tmp/usr/share/doc/aseba:
changelog  changelog.debian  copyright  example-code  example-scenarios

debian/tmp/usr/share/doc/aseba/example-code:

debian/tmp/usr/share/doc/aseba/example-scenarios:

```
But, I don't know how to solve this problem, any help is welcome

stephanemagnenat, 2018-10-15:
It seems the structure refactor introduced in 23366dafbdbbd3c9b02d951f38ac0a350e3c0dc8 broke the Debian packages. @cor3ntin could you please fix that as you change the structure?

crustyourmind, 2018-10-15:
OK thx for having to look at my problem.
I'll wait the fix and retry later
thank you very much

cor3ntin, 2018-10-16:
@crustyourmind does it work if you try to build the package from the release-1.6x branch?

crustyourmind, 2018-10-16:
First, thx for the help ;)
No sorry  it doesn't work with this branch. I used the  1.6.9, latest download from git

```
git clone --recursive https://github.com/aseba-community/aseba.git

cat VERSION.txt
#Version of aseba. Must be of the form <MAJOR>.<MINOR>.<PATCH>
1.6.9
```

cor3ntin, 2018-10-16:
do `git checkout release-1.6.x` after the git clone.
Let me know if that helps

crustyourmind, 2018-10-16:
OK If I well understand
1/ I download the source from git 
2/ I go to the aseba dir and I pass your command 
3/ I rebuild the package

OK I'll do that during the lunch, I'm at work at this time and I haven't got my laptop

stephanemagnenat, 2018-10-16:
@cor3ntin the problem in master is that the `debian/rules` file has not been updated with the new paths introduced in 23366da. For example, `targets/playground/examples/*.png` should be changed to `aseba/targets/playground/examples/*.png`.

cor3ntin, 2018-10-16:
@stephanemagnenat Should be easy enough to fix, I will take a look!

stephanemagnenat, 2018-10-16:
@cor3ntin thanks!

crustyourmind, 2018-10-19:
thx ;) I'll try to repack aseba asap

crustyourmind, 2018-10-19:
Is it possible to re open this issues ?
An another  problem appears.

Here a extract of the out of this command **debuild -i -us -uc -b**  
===============
```
cp debian/README.Debian debian/tmp/usr/share/doc/aseba/
dpkg-shlibdeps debian/tmp/usr/bin/*
dpkg-shlibdeps: avertissement: cannot find library liblibavahi.so needed by debian/tmp/usr/bin/asebaplayground (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
dpkg-shlibdeps: avertissement: cannot find library liblibavahi.so needed by debian/tmp/usr/bin/rendervplblocks (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
dpkg-shlibdeps: avertissement: cannot find library liblibavahi.so needed by debian/tmp/usr/bin/asebastudio (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
dpkg-shlibdeps: avertissement: cannot find library liblibavahi.so needed by debian/tmp/usr/bin/asebahttp (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
dpkg-shlibdeps: avertissement: cannot find library liblibavahi.so needed by debian/tmp/usr/bin/asebaswitch (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
```
===============

 cannot find library liblibavahi.so -> I think there is a path name problem (probably a missing  slash)

So I've got a aseba.deb file, but don't work...

Sorry for the trouble...

cor3ntin, 2018-10-20:
@stephanemagnenat I'm rather busy in the coming weeks, so I can't guarantee I'll have the time to really look at it.
A quick solution may be to add `-DBUILD_SHARED_LIBS=OFF` to `rules. Which is what I do at mobsya, make things much simpler for everyone.
What do you think?

crustyourmind, 2018-10-20:
Hi
I've modified the file debian/rule with the build option -DBUILD_SHARED_LIBS=OFF
The compilation works fine and a aseba-XX-.deb package is well created.
The installation of this package runs fine.
But asebastudio VPL crashes when we clic on an action button...

stephanemagnenat, 2018-10-21:
thanks @cor3ntin for the suggestion and @crustyourmind for the patch.

@crustyourmind could the crash be similat to #884? @cor3ntin did you experience similar problem on your side?

cor3ntin, 2018-10-21:
@stephanemagnenat no sorry - some spurious issues with tab creation thought, might be related, but probably not https://github.com/Mobsya/aseba/commit/c7d7a9b12175b63e63a11d76a178b3d733a881ea

I'm in the process of modifying studio / vpl rather heavily though. if I notice crashes with back-portable fixes, I will let you know!

CyrilleBiot, 2018-10-24:
Hello
Sorry for the delay but I think something had changed on the git (no more compile.Linux.md but a aseba/docs/en/development/building.rst file).
So I've rebuild the totality of the git and the build works fine. No problem. But if I'll run asestudio, always the same error when I click on a image on the right.
I 've got this error on my term 
```
$ /home/ragnarok/compilation/git/aseba/build/bin/asebastudio 
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
terminate called after throwing an instance of 'QUnhandledException'
  what():  std::exception
Abandon

```

CyrilleBiot, 2018-10-25:
Hi I just finished to compile aseba under Linux mint 18 amd 64 better than debian SID
No problem of compilation. All executables files are in```
  ls /home/libres09/git/aseba/build/bin/
asebabotspeak   asebaeventlogger  asebamassloader  aseba-qt-gui  asebatargetadvertise       aseba-test-invalid-utf8   thymioupgrader 
asebachallenge  asebaexec         asebamedulla     asebarec      asebatargetlist-dashelhub  aseba-test-msg            thymiovpl
asebacmd        asebahttp         asebaplay        asebashell    asebatargetlist-qt         aseba-test-natives-count  thymiownetconfig
asebadummynode  asebahttp2        asebaplayground  asebastudio   asebatargetlist-thread     aseba-test-simulator      thymiownetconfig-cli
asebadump       asebajoy          aseba-qt-dbus    asebaswitch   asebatest                  rendervplblocks           tst_compiler_utf8
```

And under mint 18 , aseba studio doesn't crash

mbonani, 2018-10-25:
normally blocky disapear only for OSx and windows should be there in Linux base system

CyrilleBiot, 2018-10-25:
@mbonani : no blockly is not more present in the last compilation af asebastudio
It's not really a problem, I can install http://blockly4thymio.net//index.php/essayer.html
but it was easier ama to have all this programs in a one package

CyrilleBiot, 2018-10-26:
So happy, I've found the bug ;)
I  set an  additional locate in /etc/locale.gen
My file contained only fr_FR.UTF-8 UTF-8.
I've added en_US. UTF-8 UTF-8 entry and regenerate the locale with the commande **locale-gen** 
I've restarted asebastudio and no more crash.
Fix for me ;)

CyrilleBiot, 2018-10-26:
OK I think you can fix this thread. I've just finsihed to compil a debian package for debian SID amd64
You'll find it here : https://github.com/CyrilleBiot/aseba/tree/master/debian-packages
or here
https://www.cbiot.fr/aseba/
Thx for the help ;)

PR #886 (2018-10-19): Fix playground paths in debian package
Fixes #885
commit 4c8a04d7 (2018-10-19): Fix playground paths in debian package

Fixes #885