aseba-community/aseba #891: asebastudio crashes on fedora

nielsk, 2019-03-04:
I built aseba on Fedora 29 and it built and installed but when I call asebastudio it crashes:
[1]    14364 segmentation fault (core dumped)  asebastudio

I started it with strace and get this:
```
nik@netcat ~ strace asebastudio 
execve("/usr/local/bin/asebastudio", ["asebastudio"], 0x7ffe028fa4d0 /* 106 vars */) = 0
brk(NULL)                               = 0x1a5d000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffc44e7e850) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=262224, ...}) = 0
mmap(NULL, 262224, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fb1bdd4d000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libprotobuf.so.15", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360l\r\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=3172952, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb1bdd4b000
mmap(NULL, 3100120, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb1bda56000
mprotect(0x7fb1bdb23000, 2195456, PROT_NONE) = 0
mmap(0x7fb1bdb23000, 1634304, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xcd000) = 0x7fb1bdb23000
mmap(0x7fb1bdcb2000, 557056, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25c000) = 0x7fb1bdcb2000
mmap(0x7fb1bdd3b000, 61440, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2e4000) = 0x7fb1bdd3b000
mmap(0x7fb1bdd4a000, 3544, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb1bdd4a000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libqwt.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@*\7\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1435768, ...}) = 0
mmap(NULL, 1368832, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb1bd907000
mprotect(0x7fb1bd972000, 876544, PROT_NONE) = 0
mmap(0x7fb1bd972000, 671744, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6b000) = 0x7fb1bd972000
mmap(0x7fb1bda16000, 200704, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10f000) = 0x7fb1bda16000
mmap(0x7fb1bda48000, 57344, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x140000) = 0x7fb1bda48000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libQt5Help.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\274\1\0\0\0\0\0"..., 832) = 832
lseek(3, 504736, SEEK_SET)              = 504736
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
fstat(3, {st_mode=S_IFREG|0755, st_size=536744, ...}) = 0
lseek(3, 504736, SEEK_SET)              = 504736
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
mmap(NULL, 524336, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb1bd886000
mmap(0x7fb1bd89f000, 335872, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19000) = 0x7fb1bd89f000
mmap(0x7fb1bd8f1000, 69632, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6b000) = 0x7fb1bd8f1000
mmap(0x7fb1bd902000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x7b000) = 0x7fb1bd902000
mmap(0x7fb1bd906000, 48, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb1bd906000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libQt5Svg.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\270\1\0\0\0\0\0"..., 832) = 832
lseek(3, 341664, SEEK_SET)              = 341664
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
fstat(3, {st_mode=S_IFREG|0755, st_size=364264, ...}) = 0
lseek(3, 341664, SEEK_SET)              = 341664
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
mmap(NULL, 360504, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb1bd82d000
mprotect(0x7fb1bd845000, 249856, PROT_NONE) = 0
mmap(0x7fb1bd845000, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7fb1bd845000
mmap(0x7fb1bd876000, 45056, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x49000) = 0x7fb1bd876000
mmap(0x7fb1bd882000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x54000) = 0x7fb1bd882000
mmap(0x7fb1bd885000, 56, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb1bd885000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libQt5Concurrent.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p4\0\0\0\0\0\0"..., 832) = 832
lseek(3, 27480, SEEK_SET)               = 27480
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
fstat(3, {st_mode=S_IFREG|0755, st_size=33032, ...}) = 0
lseek(3, 27480, SEEK_SET)               = 27480
read(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32) = 32
mmap(NULL, 32792, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb1bd824000
mmap(0x7fb1bd827000, 12288, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7fb1bd827000
[... 930 lines cut ...]
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb1b8180000
mprotect(0x7fb1bdd3b000, 57344, PROT_READ) = 0
mprotect(0x7ae000, 53248, PROT_READ)    = 0
mprotect(0x7fb1bddb8000, 4096, PROT_READ) = 0
munmap(0x7fb1bdd4d000, 262224)          = 0
set_tid_address(0x7fb1b8183c10)         = 14540
set_robust_list(0x7fb1b8183c20, 24)     = 0
rt_sigaction(SIGRTMIN, {sa_handler=0x7fb1bc571c50, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0x7fb1bc57e030}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0x7fb1bc571ce0, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0x7fb1bc57e030}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
brk(NULL)                               = 0x1a5d000
brk(0x1a7e000)                          = 0x1a7e000
statfs("/sys/fs/selinux", {f_type=SELINUX_MAGIC, f_bsize=4096, f_blocks=0, f_bfree=0, f_bavail=0, f_files=0, f_ffree=0, f_fsid={val=[0, 0]}, f_namelen=255, f_frsize=4096, f_flags=ST_VALID|ST_RELATIME}) = 0
statfs("/sys/fs/selinux", {f_type=SELINUX_MAGIC, f_bsize=4096, f_blocks=0, f_bfree=0, f_bavail=0, f_files=0, f_ffree=0, f_fsid={val=[0, 0]}, f_namelen=255, f_frsize=4096, f_flags=ST_VALID|ST_RELATIME}) = 0
access("/etc/selinux/config", F_OK)     = 0
access("/etc/system-fips", F_OK)        = -1 ENOENT (No such file or directory)
futex(0x7fb1b8b620e8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fb1b8b56fb8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fb1b8b56fb8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
access("/etc/system-fips", F_OK)        = -1 ENOENT (No such file or directory)
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0x7fb1bc8a2a60} ---
+++ killed by SIGSEGV (core dumped) +++
[1]    14537 segmentation fault (core dumped)  strace asebastudio
```

mbonani, 2019-03-11:
Please consider that we have fork this project on https://github.com/mobsya/aseba. We are actually in a big release change and will provide soon new Linux build using flatpack (for master branch)
The last stable release is on branch https://github.com/Mobsya/aseba/tree/release-1.6.x

nielsk, 2019-04-13:
Sorry for coming back this late. In the meantime I installed a VM with ubuntu where it works. I will try out now the mobsya-repo and report further problems there. I close this.
Thanks a lot for your help.

closed 2019-04-13 by nielsk