aseba-community/aseba #893: aseba compilation issue on raspberry

eziosoma, 2019-06-14:
Hello,
I have tried to generate the package of Aseba on Raspberrry An I have got a ot of compilation errors.
Looking for the cause i have found a similar [bug](https://bugs.debian.org/895499). The errors are very similar and are relate to the package aseba-enky.
The root cause is:
" armel/armhf (and arm64 in Ubuntu) Qt5 is compiled with OpenGL ES instead of OpenGL.".

Can this apply also to aseba on raspberry problems?

mbonani, 2019-06-14:
We migrate for Thymo our repo here: https://github.com/Mobsya/aseba
 perhaps it is easier to compile for Rasberry?

eziosoma, 2019-06-16:
Hello, I have done a test on ubuntu before to try with Raspberry, and I
have got the following errors.

Trying under ubuntu 18.10, command:

cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF ..

got the following errors:

-- Git executable found
-- libudev found
Using dashel from /home/ezio/aseba/aseba/dashel
-- Could NOT find Boost
Using Enki from /home/ezio/aseba/aseba/enki
-- Version: 5.3.1
-- Build type: Release
-- CXX_STANDARD: 17
CMake Warning (dev) at third_party/quazip/quazip/CMakeLists.txt:28
(target_link_libraries):
  Policy CMP0023 is not set: Plain and keyword target_link_libraries
  signatures cannot be mixed.  Run "cmake --help-policy CMP0023" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The keyword signature for target_link_libraries has already been used with
  the target "quazip5".  All uses of target_link_libraries with a target
  should be either all-keyword or all-plain.

  The uses of the keyword signature are here:

   * third_party/quazip/quazip/CMakeLists.txt:21 (target_link_libraries)

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at third_party/quazip/quazip/CMakeLists.txt:29
(target_link_libraries):
  Policy CMP0023 is not set: Plain and keyword target_link_libraries
  signatures cannot be mixed.  Run "cmake --help-policy CMP0023" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The keyword signature for target_link_libraries has already been used with
  the target "quazip_static".  All uses of target_link_libraries with a
  target should be either all-keyword or all-plain.

  The uses of the keyword signature are here:

   * third_party/quazip/quazip/CMakeLists.txt:20 (target_link_libraries)

This warning is for project developers.  Use -Wno-dev to suppress it.

-- [range-v3]: C++ std=11
-- Found ZeroConf in /usr/include;/usr/lib/x86_64-linux-gnu/libdns_sd.so
Using avahi as backend
asebamassloader HAS_BUNDLE-NOTFOUND
thymiownetconfig HAS_BUNDLE-NOTFOUND
CMake Error at aseba/clients/qtcommon/CMakeLists.txt:19
(target_link_libraries):
  Object library target "asebaqtcommon" may not link to anything.

-- Configuring incomplete, errors occurred!
See also "/home/ezio/aseba/aseba/build/CMakeFiles/CMakeOutput.log".
See also "/home/ezio/aseba/aseba/build/CMakeFiles/CMakeError.log".

eziosoma, 2019-06-16:
the error and log file are attached.