aseba-community/aseba #895: Update of Aseba protocole

buragohf, 2019-12-08:
Hi, i'm running aseba-git available in AUR  (for Arch, Manjaro....) and i getting a protocole issue with aseba.
When connecting my Thymio to my computer and launching "Aseba studio for Thymio", the application is well opening but i'm getting the following error message : "Aseba Studio utilise un protocole plus ancien (5) que le noeud thymio-II (9), veuillez mettre à jour Aseba Studio."

meaning "Aseba studio use an older protocole (5) than the node thymio-II(9), please update aseba Studio"

my current installed version of aseba studio is version 1.6.9.

could you please explain how to fix it  using linux manjaro?

thank you very much in advance for the time you spend on this project.

burago

THS-on, 2019-12-09:
There are two versions of Aseba and the AUR version currently uses the one from this repository.  
Mobsya has forked Aseba and you might need their Version. You can try the Flatpack from here: https://www.thymio.org/help/linux-installation/

If I have time I'll make a new AUR package for the Mobsya version.

buragohf, 2019-12-14:
Hi, thank you for your quick reply reply. i will try the flatpack this weekend.
for sure, i would really appreciate a AUR package. it would be much easier.
thanks again

buragohf, 2019-12-15:
Hello, i succeed to install the application but when i launch it (flatpak run org.mobsya.ThymioSuite)
i'm getting a black windows. 
if i launch it from the shell, i can see, i parallel of the black windows Thymio suite, the following message repeating
"...
ThymioDeviceManagerClientEndpoint::handleIncommingMessage NodesChanged
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping"

could you please support or explain me where i should adress my questions?
burago

mbonani, 2019-12-16:
Hello, if you are use to github, the new repo is here: https://github.com/Mobsya/aseba/ 
You have also installation information here: https://www.thymio.org/help/linux-installation/
The udev rules have to  be done for making Thymio Suite works. 
Is your system 32bits or 64bits?

buragohf, 2019-12-16:
Hi, my OS is manjaro x86_64.

i did the udev configuration before by making
$ sudo nano /etc/udev/rules.d/99-mobsya.rules
to paste
SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000a",
MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000c",
MODE="0666"
Then I saved

$ sudo udevadm control --reload-rules

is there something i missed due to my 64bits machines?
my manjaro is using xfce environnement. could it bring issues?

best regards,
burago

THS-on, 2019-12-16:
@buragohf have you added your user to the `uucp`  group?  
You can do this with:
```
sudo usermod -a -G uucp $USER
```

buragohf, 2019-12-16:
No i didn't do it previous so i just did it now....
unfortunatelly, no difference.

i provide you here below what i can see in the shell

[user@machine ~]$ flatpak run org.mobsya.ThymioSuite
Gtk-Message: 14:34:16.582: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:34:16.611: Failed to load module "canberra-gtk-module"
Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
QApplication: invalid style override passed, ignoring it.
    Available styles: Adwaita-Dark, Adwaita, Breeze, Windows, Fusion
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
Searching for  "thymio-device-manager"
"thymio-device-manager" found:  "/app/bin/thymio-device-manager"
thymio-device-manager starting
Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created.
"fr_FR" "launcher_" ":/translations"
"fr_FR" "qtdeclarative_" "/usr/translations"
Didn't load translation "qtdeclarative_"
"fr_FR" "qtbase_" "/usr/translations"
Didn't load translation "qtbase_"
"fr_FR" "qtwebengine_" "/usr/translations"
Didn't load translation "qtwebengine_"
Searching for  "asebaplayground"
"asebaplayground" found:  "/app/bin/asebaplayground"
thymio-device-manager started
zeroconf: ThymioDeviceManagerClient::onServiceAdded:  "Zeroconf Service: Thymio Device Manager on machine @ machine.local (192.168.xxx.xxx:42177)"
zeroconf: ThymioDeviceManagerClient::onServiceAdded: id:  QUuid("{7c7105e7-67........5}")
ThymioDeviceManagerClientEndpoint::handleIncommingMessage ConnectionHandshake
ThymioDeviceManagerClientEndpoint::handleIncommingMessage NodesChanged
zeroconf: ThymioDeviceManagerClient::onServiceAdded:  "Zeroconf Service: Thymio Device Manager on machine @ machine.local (192.168.xxx.xxx:42177)"
zeroconf: ThymioDeviceManagerClient::onServiceAdded: id:  QUuid("{7c7105e7.......1025}")
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping

mbonani, 2019-12-16:
For me if you still have a black windows you probably have problems with OpenGL. ThymioDeviceManager traces are looking ok. Perhaps problem come also from canberra-gtk-module, I do not really like the error message at beginning, lot of post on internet seems to solve this issue.

buragohf, 2019-12-16:
thank you, i have the same feeling regarding OpenGL and I tried to
reinstall my graphic card driver but without any success
i didn't succeed to solve the canberra-gtk-module issue but it seems
related to the sound
 so it's should not be mandatory.
i feel a bit blocked and i don't know where to look at.

thank you again for your support.