aseba-community/aseba #898: Building a playground

maryam6262, 2024-12-28:
Hello aseba. I would  be thank if you help me createing a playground with an image inside it. something like the playgrounds in example folder.
Thanks in advance

FrancescoMondada, 2025-01-06:
Have you looked to https://github.com/Mobsya/aseba/blob/master/aseba/targets/playground/examples/HowToBuildAPlayground.playground ? You have a ground and an image as texture...

maryam6262, 2025-01-06:
Thank you so much. Yes I have seen this example but if I'm not mistaken the picture file should be next to the aseba file for showing that as background. But I want the picture be embed in playground file not next to it. Thanks a lot in advance

davidjsherman, 2025-01-07:
If I understand your question, you would like a single `.playground` file, like [thymio-discovery-maisondailleurs.playground](https://github.com/Mobsya/aseba/blob/master/aseba/targets/playground/examples/thymio-discovery-maisondailleurs.playground). A combined playground file is a ZIP archive, with no directories, where the playground file is named `world.xml`.

Probably you have already done steps 1–3 and you just need to do step 4:

1. Name your XML playground definition **`world.xml`**.
2. Place the PNG image file in the same directory. Use any name you like.
3. Refer to the image by name in the `groundTexture` attribute of the `<world>` tag in `world.xml`.
4. **Create a ZIP archive of the two files, using `.playground` as the suffix instead of `.zip`.**

maryam6262, 2025-01-07:
> If I understand your question, you would like a single `.playground` file, like [...]

Thank you alot. That's exactly thing I want. Appreciate you

maryam6262, 2025-01-27:
Hello everyone. May you please tell me where can I share my Thymio playgrounds with others?

FrancescoMondada, 2025-02-06:
Could be your playgrounds a contribution in the examples of playground of this package?

maryam6262, 2025-02-16:
Hello, I will send you some of them. Anyway, it would be nice if there was
a forum to share the new playgrounds for kids who are learning Thymio. My
students love my new playgrounds and I want to share them with others.

FrancescoMondada, 2025-02-18:
Hello, we do not have a forum well suited for this, sorry. The plaground could be shared by mobsya (you can contact them at info@mobsya.org ) or putting them in the original package.